mesh-iit/ergocub-software #5: Add controlboard ini files for commanding the hands mk5

Nicogene, 2022-12-21:
For doing this we have to take account of the coupling of the joints, and check the limits set in the csv files

mfussi66, 2022-12-21:
The left hand had already the coupling coded up: https://github.com/robotology/gazebo-yarp-plugins/pull/624

Nicogene, 2023-01-11:
Today I tried to write an ini file like the one we had for the lower arm with the hand mk4 but unfortunately since `thumb_rotate` has been removed, adjust the ini removing the joint is not sufficient because that [HandCouplingHandler](https://github.com/robotology/gazebo-yarp-plugins/blob/b9c18f46107375efc90e9000e0167d1a7c8b7054/plugins/controlboard/include/yarp/dev/ControlBoardDriverCoupling.h#L200) is expecting that that joint exists.

We should create the couplinghandlermk5

PR #26 (2023-01-13): ergoCubGazeboV1: add controlboards for commanding the fingers
This PR adds the files for commanding the fingers of the hands.

It needs https://github.com/robotology/gazebo-yarp-plugins/pull/641.

It fixes #5 

The limits of the fingers joints probably will have to be revised