mesh-iit/ergocub-software #7: Check joints orientations

Nicogene, 2022-12-21:
When generating the urdf the orientation is assigned arbitrary, then probably we will need to put the joints inverted under this list:
```
reverseRotationAxis:
...
```

mfussi66, 2022-12-21:
I noticed in [this issue](https://github.com/icub-tech-iit/ergocub-software/issues/8) that the right wrist yaw rotates counterclockwise when the setpoint is negative, so it's the opposite of the expected behaviour. The wrists need to be mirrored, and the direction is the following for both left and right:
   - positive yaw: palm towards the robot
   - positive roll: palm towards the robot
   - positive pitch: hand tilts forward

mfussi66, 2022-12-23:
I wanted to play around with the model and I performed the change for curiosity:

https://user-images.githubusercontent.com/38140169/209373652-8c1bee5c-7bfd-4671-8321-9aba574b2dfd.mp4

It works as expected, @Nicogene I can open the PR if you haven't made the change locally.

Nicogene, 2022-12-27:
Here: https://github.com/icub-tech-iit/ergocub-software/commit/874f75e5623707bb925ec4f273bc21daa765244c I fixed some joints orientations

Nicogene, 2022-12-27:
Keeping open because we didn't checked (and eventually) fixed all joints. In this sense unit test (#6) could help

Nicogene, 2023-01-09:
With #22 I managed to fix some joints' orientations but I have these errors that I cannot fix simply inverting the joints:
(NOTE: the expected has been taken from icub3 model)

```
r_shoulder_yaw got direction of  x -0.116648 y 0.227771 z 0.966702 instead of expected  x -0.0713662 y 0.0619303 z 0.995526
r_elbow got direction of  x 0.250563 y -0.935113 z 0.250563 instead of expected  x 0.267012 y -0.960459 z 0.0788901
l_shoulder_yaw got direction of  x -0.116648 y -0.227771 z 0.966702 instead of expected  x 0.0713662 y 0.0619303 z -0.995526
l_elbow got direction of  x -0.250563 y -0.935113 z -0.250563 instead of expected  x -0.267012 y -0.960459 z -0.0788901
```

If these joints are identical between ergocub and icub3 it means there is a problem in the simmechanics, otherwise we can just change the test

cc @traversaro @mfussi66 @pattacini @salvi-mattia @Mick3Lozzo

salvi-mattia, 2023-01-10:
togheter with @Nicogene 
we investigated the differencies found in the shoulder yaw axis and elbow axis between icub3 model and ecub model.

we can see that the mutual orientation between shoulder yaw axis and root frame is the same in both icub3 and ecub1
[the distance is different because ecub is taller]

[image]

the differences noticed by @Nicogene is due to a different positioning of the two simulation models [mechanism]

[image]

it is therefore normal that each axis below the shoulder roll has a different positioning 
--> shoulder_yaw, elbow, pronosupination, wrist roll & pitch and every axis in the hand

pattacini, 2023-01-10:
Nice analysis!

Probably, this could be solved at the test level by computing the required quantities (distances, angles...) relative to the root.

closed 2023-01-13 by Nicogene