mesh-iit/ergocub-software #8: Wrist yaw goes in HF when moved with the motorgui

GiulioRomualdi, 2022-12-21:
Hi all first of all thank you for the model!

I tried to import the model into gazebo and move some of the joints with the motorgui. I noticed that the Wrist yaw goes in HF when as soon as I move it.

Here a video

https://user-images.githubusercontent.com/16744101/208940803-348f81ec-d261-4f26-b60f-54d840448b19.mp4

cc @Nicogene @traversaro

fiorisi, 2022-12-21:
We might have to increase the inertia of the link as done for the single forearm: https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/components/left_hand_mk3/simmechanics2urdf_configfile.yaml#L42

Another possibility is to use the `implicitSpringDamper`.

traversaro, 2022-12-21:
Probably the usual problems induced by the combination of small inertia and the default physics settings of Gazebo Classic ?

Nicogene, 2022-12-22:
> Probably the usual problems induced by the combination of small inertia and the default physics settings of Gazebo Classic ?

In theory, I assigned to all the links of the hands 0.01 inertia but maybe I forgot some of them

PR #17 (2022-12-27): Fixes wrist HF
It was due to hw and cb limits mismatch for copy-paste error, fixes #8
commit 4b88ec81 (2022-12-27): Fixes wrist HF

It was due to hw and cb limits mismatch for copy-paste error, fixes #8
closed 2022-12-27 by Nicogene