mesh-iit/ergocub-software #10: Incompatibility between PTC Creo 9 and Simscape Multibody Link

pattacini, 2022-12-22:
See discussion in #81.

Reported issue in https://www.mathworks.com/matlabcentral/answers/1879607-incompatibility-between-ptc-creo-9-and-simscape-multibody-link and filed a bug report with case number **05918242**.

cc @icub-tech-iit/silo-mech @icub-tech-iit/fix @maggia80

mfussi66, 2022-12-22:
Here is the communication history of the bug report so far:

## Mathworks:

>  Dear Mattia,
> I  am Allan from MathWorks Technical Support. I am writing to you in  reference to your Technical Support Case #05918242 'Incompatibility  between Simscape Link and PTC Creo 9'.I understand that you are  experiencing issues when importing revolute joints using Simscape  Multibody Link with PCT Creo 9.0. I also see that you have been able  achieve this workflow using Creo 7.2, but when you import using Creo 9.0  the expected XML for the revolute joints is absent.You also mention that there are no references in the XML to the STEP or STL files associated with your project.I am working closely with my colleagues to investigate this for you. I will keep you informed with any updates.While we investigate, are you still able to use Creo 7.6, or is this a blocking issue for you?I look forward to hearing back from you.
> Sincerely,
> Allan Petrie
> MathWorks Technical Support 
> Use the following link to view and update your Case online:http://www.mathworks.com/support/servicerequests/index.htmlPlease  preserve the Reference ID in any further correspondence on this query.  This will allow our systems to automatically assign your reply to the  appropriate Case.
> **ref:_00Di0Ha1u._5003q1Z62a3:ref**

## Us:

> Dear Allan,
>  Thank you for looking into it. Sadly yes, it is a blocking issue, because our mechanical engineering department upgraded to Creo 9 and it is not possible to revert to previous versions.

## 🟢 New reply from Mathworks 🟢

> Dear Mattia,
> 
> Thank you for clarifying this point.
> 
> Our Development Team have been made aware of the issue that you are facing and are investigating a solution.
> 
> I have ensured they understand this is a blocking issue for you, however, this investigation may take some time - especially as many colleagues will be out of office until the new year.
> 
> I will keep you up to date with any updates and information in the meantime.
> 
> Best regards,
> Allan Petrie
> MathWorks Technical Support
> 
> This is in reference to your Technical Support Case #05918242 'Incompatibility between Simscape Link and PTC Creo 9'.
> 
> Use the following link to view and update your Case online:
> http://www.mathworks.com/support/servicerequests/index.html
> 
> Please preserve the Reference ID in any further correspondence on this query. This will allow our systems to automatically assign your reply to the appropriate Case.

mfussi66, 2022-12-22:
The current state of the bug report is that the development team has been informed of the bug, and they will keep us updated after confirming that the incompatibility is blocking for us.

mfussi66, 2023-01-05:
Got a reply from Mathworks, in which they admitted that Creo 9 introduced breaking changes and they will need to fix the plug-in. It will land in a future release or update, I wasn't informed of a timeframe, but I'll be notified when the fix is released. 

However, it seems that  `simmechanics_to_urdf` will need to be retired.

pattacini, 2023-01-12:
Closing for now.

closed 2023-01-12 by pattacini