mesh-iit/ergocub-software #12: Missing SCSYS in soles

Nicogene, 2022-12-23:
We are missing both `SCSYS_R_SOLE` and `SCSYS_L_SOLE` in the simmechanics xml

See https://github.com/search?q=repo%3Aicub-tech-iit%2Fergocub-software%20SCSYS_R_SOLE&type=code

cc @fiorisi @traversaro @pattacini @Lawproto @Mick3Lozzo @salvi-mattia

Mick3Lozzo, 2022-12-23:
[image]

@Nicogene they are present in the shrink-wrapped Creo 9 assembly....

[image]
 
.... but NOT in the shrink-wrapped Creo 7 assembly. That should not be possible, because we used the Creo 9 assembly as a source for the Creo 7 assembly. 💀💀

Well, we have to redo a couple of shrinkwraps. Maybe in some of the many passages back and fort we did, we lost something.

Lawproto, 2022-12-23:
@Mick3Lozzo I exported a .neu of `SIM_ECUB_L_FOOT_REAR` on the fly and opened it in Creo7. I can see the SCSYS:

[image]

I suggest you recreate the two neutrals separately, overwrite the Creo 7 links and fix the assembly. Remember to reassign mass with the external file.

<details>
Hohoho! 🎅 
</details>

Mick3Lozzo, 2022-12-23:
[image]

Added missing SCSYS @Nicogene.

Nicogene, 2022-12-27:
The part on the yml files should be already configured for exporting the frames, as soon as @mfussi66 can regenerate the simmechanics XML we should have the frames in the urdf

Nicogene, 2022-12-27:
Added in #20 and #21, closing!

closed 2022-12-27 by Nicogene