mesh-iit/ergocub-software #23: r_sole and l_sole are not parallel

Nicogene, 2023-01-09:
#22 added the unit tests that checks if `l_sole` and `r_sole` are parallel.

It gives the following error:
```
l_sole_y is 0.07435, while r_sole_y is -0.07445 while they should be simmetric (diff : 0.0001 )
```

`sole_height` and `sole_x` seems fine instead

cc @salvi-mattia @traversaro @mfussi66 @pattacini @Mick3Lozzo

salvi-mattia, 2023-01-09:
in cad model they r parallel

[image]

and they are parallel in the SW mechanism

[image]

where does this misalignement came from??

Nicogene, 2023-01-09:
I tried to compute from code the same transform as:

```
    iDynTree::Transform l_sole_H_r_sole = comp.getRelativeTransform(l_sole,r_sole);
    std::cerr<<"l_sole - r_sole "<<l_sole_H_r_sole.getPosition().toString()<<std::endl;
```

And I got:

```
l_sole - r_sole -5e-05 -0.1488 0
```
That is the same as the cad.

Since we are computing the alignment of the frames using the two transforms `root_link_H_r_sole` and `root_link_H_l_sole`, maybe the `root_link` frame is not perfectly centered?

cc @fiorisi @traversaro @Mick3Lozzo @salvi-mattia

salvi-mattia, 2023-01-09:
the two soles' CSYS are off center
@Mick3Lozzo is investigating

[image]

[image]

Mick3Lozzo, 2023-01-09:
The problem is the placement of the upper body on the lower body, is merely a CAD assembling problem.

[image]

⬆️Between the bright green supports of the torso pitch motor and the dark green plate housing the torso yay motor, there is a 0.1 mm gap in assembling the components without interference problems due to the tolerance chain.

[image]

⬆️  This gap should be equally distributed between the joint's left and right sides of the joint but is all on one side.  

[image]

⬆️  This `coincident` should be `offset 0.05 mm`.

I have corrected the misalignment, and now the robot is fully symmetrical again ⬇️.

[image]

[image]

We must correct some shrinkwrap, probably the root link and the other links directly constrained to it.

@fiorisi @traversaro @mfussi66  @salvi-mattia @Nicogene

Mick3Lozzo, 2023-01-10:
> We must correct some shrinkwraps, probably the root link and the other links directly constrained to it.
> 
> @fiorisi @traversaro @mfussi66 @salvi-mattia @Nicogene

Done, both in Creo 7 and in Creo 9 shrink-wraps, we can generate e new URDF @Nicogene.

cc @fiorisi @traversaro @mfussi66 @salvi-mattia

Mick3Lozzo, 2023-01-12:
Resolved, and as a bonus, I resolved some other inconsistencies I found on the shoulders.

[image]

[image]

[image]

We can re-generate the URDF again (until the next bug).

@fiorisi @traversaro @mfussi66 @salvi-mattia @Nicogene

closed 2023-01-13 by Nicogene