mesh-iit/ergocub-software #28: Wrong exportation of `r_knee` from CREO to Simmechanics

pattacini, 2023-01-13:
Right now I had to manually fix the simmechanics XML for the `r_knee` exported as weld, and the missing stl for `l_wrist_1` link

- [`50237ab` (#27)](https://github.com/icub-tech-iit/ergocub-software/pull/27/commits/50237abc7f8f8c2ba0cd817120369d3f772e42e4)
- [`0200cab` (#27)](https://github.com/icub-tech-iit/ergocub-software/pull/27/commits/0200cab145a85586197cff5da7ae2342e7ef1083)

Shall we keep track in a separate issue of these glitches in the exportation from creo?

@Lawproto @mfussi66 @Mick3Lozzo @pattacini @salvi-mattia @fiorisi

_Originally posted by @Nicogene in https://github.com/icub-tech-iit/ergocub-software/issues/27#issuecomment-1381604207_

mfussi66, 2023-01-13:
To add a bit of details:
- Regarding the left wrist STL, we get the following error:
```
Unable to triangulate part SIM_ECUB_L_WRIST_1.  Aborting output.
Warning: The stl file for SIM_ECUB_L_WRIST_1 could not be created.
```
The options in the Simscape Plugin do not seem to be able to fix the issue, maybe we can try to lower the shrinkwrap quality or play around with the stl export options? (cc @lawproto)
- Regarding the right knee, there is no warning or error at all during the export process

fiorisi, 2023-01-13:
> The options in the Simscape Plugin do not seem to be able to fix the issue, maybe we can try to lower the shrinkwrap quality or play around with the stl export options? (cc @Lawproto)

Yes, usually changing the shrinkwrap quality solves the problem.

salvi-mattia, 2023-01-13:
there is a span in the quality level in the guidelines [it seems me 7-9]
it is better to use the lowest that give the result rather than go straight for the highest

Lawproto, 2023-01-13:
> it is better to use the lowest that give the result rather than go straight for the highest

Did not know this prior to tackling the pre-URDF sim model experience, I went for the highest whenever possible instead. Thanks for the suggestion.

FYI I noticed that the same quality number seem to give pretty different mesh result for very similar or even equal links. Felt kinda odd, to be honest. This is a copmarison between `l_wrist_1` and `r_wrist_1`.
.
[image]

<details>Guess which is on the left and which is on the right of the image 😃</details>

salvi-mattia, 2023-01-13:
> Did not know this prior to tackling the pre-URDF sim model experience, I went for the highest whenever possible instead. Thanks for the suggestion.
> 

you live, you learn

> FYI I noticed that the same quality number seem to give pretty different mesh result for very similar or even equal links. Felt kinda odd, to be honest. This is a copmarison between `l_wrist_1` and `r_wrist_1`. . [image]
> Guess which is on the left and which is on the right of the image

yes, it happens

Lawproto, 2023-01-13:
With the commit above, I did all the necessary steps to recreate a new `l_wrist_1` model for the Creo 7 ecub sim model. I chose quality 6, which is the lowest contemplated by our guidelines.
@mfussi66 please let me know if this solves your problem.

mfussi66, 2023-01-13:
The wrist seems fine now! See the meshlab screenshot of the opened STL:
[image]

The knee issue is persisting.

Nicogene, 2023-08-07:
Since we decided to not generate anymore `ergoCub` 1.0 urdf using `simmechanic_to_urdf` and since it seems to not happen when using `creo2urdf` we can close this issue.

closed 2023-08-07 by Nicogene