mesh-iit/ergocub-software #35: Write configuration file for launching wholebodydynamics for ergoCubGazeboV1 (simulated robot) as part of the gazebo simulation process via the gazebo_yarp_robotinterface

traversaro, 2023-01-23:
For running walking and other demos, it is necessary to estimate external forces acting on the robot, and this can be done via the `wholebodydynamics` YARP device. This YARP device can run:
* `wbd-int`. Inside the gazebo simulator, via the `gazebo_yarp_robotinterface` plugin, a bit like on the real robot wholebodydynamics is running in the main yarprobotinterface of the real robot `ergoCubSN000` (see https://github.com/robotology/robots-configuration/blob/v2.1.0/ergoCubSN000/estimators/wholebodydynamics.xml) 
* `wbd-ext` Outside of the gazebo simulator, by accessing the relevant sensor via the `multipleanalogsensorclient` device

This issue tracks the `wbd-int` option.

traversaro, 2023-01-23:
To solve this, we should add the appropriate conffile for wbd in https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/ergocub.xml .

mebbaid, 2023-01-23:
I started by adding the line 
``` 
  <!--  estimators  -->
<xi:include href="estimators/wbd_ecub_sim.xml"/>
```

at the end of https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/ergocub.xml, and then populating the corresponding `wbd_ecub_sim.xml` config file

mebbaid, 2023-02-06:
After some modifications on the estimator xml configuration file following the one set for the real robot in https://github.com/robotology/robots-configuration/blob/v2.1.0/ergoCubSN000/estimators/wholebodydynamics.xml, i am now facing this error

```
[ERROR] wholeBodyDynamics : openContactFrames : missing default/override contact for submodel composed by the links: 
[ERROR] wholeBodyDynamics : openContactFrames : r_shoulder_3
[ERROR] wholeBodyDynamics : openContactFrames : r_upperarm
[ERROR] wholeBodyDynamics : openContactFrames : r_forearm
[ERROR] wholeBodyDynamics : openContactFrames : r_wrist_1
[ERROR] wholeBodyDynamics : openContactFrames : r_wrist_2
[ERROR] wholeBodyDynamics : openContactFrames : r_hand_palm
[ERROR] wholeBodyDynamics: Problem in opening default contact frame settings.
[ERROR] |yarp.dev.PolyDriver|wholebodydynamics| Driver <wholebodydynamics> was found but could not open
[WARNING] Cannot open device wholebodydynamics
```

traversaro, 2023-02-06:
I think we need to add a default contact in `r_hand_palm`, i.e. change `r_hand`  in `r_hand_palm`  in https://github.com/robotology/robots-configuration/blob/v2.1.0/ergoCubSN000/estimators/wholebodydynamics.xml#L9 .

mebbaid, 2023-02-07:
After a f2f with @traversaro, errors related to the coupling in the hands were fixed by applying https://github.com/icub-tech-iit/ergocub-software/issues/35. We got back to this error

```
[DEBUG] wholeBodyDynamicsDevice :: number of ft sensors found in both ft + mas 16 where analog are  8  and mas are  8
[ERROR] WholeBodyDynamicsDevice::attachAll Invalid number of MAS FT sensors after remapper
```
 
which is triggered in the following lines in the `wholebodydynamics` device https://github.com/robotology/whole-body-estimators/blob/c46c1852974776038a75b7c0e542ccb2917d7c7f/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L1743-L1747.

Further inspecting the contents of both `nrMASFTSensors` and `remappedMASInterfaces.ftMultiSensors->getNrOfSixAxisForceTorqueSensors()` I get 8 MAS FT sensors and 0 MAS rempped  FT sensors respectively. This seems very strange to me

mebbaid, 2023-02-07:
It seems the same error was investigated in https://github.com/robotology/whole-body-estimators/issues/163
by @Hosam , maybe some group in the configuration is missing as per this comment
https://github.com/robotology/whole-body-estimators/issues/163#issuecomment-1268289116

mebbaid, 2023-02-09:
The xml files to run the `wholebodydynamics` device were prepared and tested with the robot in simulation. In case it is ok, I can PR adding the files to the conf directory (namely this directly of the repo https://github.com/icub-tech-iit/ergocub-software/tree/master/urdf/ergoCub/conf )

otherwise @pattacini @Nicogene feel free to let me know of a better place to commit those files. 
The same applies for #34

Nicogene, 2023-02-10:
The PR has been merged, shall we close this issue?

pattacini, 2023-02-10:
I think they aim to churn out a second PR for that.