mesh-iit/ergocub-software #36: Inconsistency in head IMU sensor name between simulated and real robot

traversaro, 2023-01-23:
While investigating on https://github.com/icub-tech-iit/ergocub-software/issues/34, we noticed a difference between the simulated robot with `YARP_ROBOT_NAME` `ergoCubGazeboV1` and the real robot with `YARP_ROBOT_NAME` `ergoCubSN000`.

### Simulated Robot `ergoCubGazeboV1`

In the simulated robot, the name of the head IMU is `head_imu_acc_1x1` : https://github.com/icub-tech-iit/ergocub-software/blob/2b6874d45531a86599190de831b3f2e6bf1676f1/urdf/simmechanics/data/ergocub/ERGOCUB_all_options.yaml#L804 . In particular, there are four sensors of different kind as part of this IMU according to the related `multipleanalogsensorsserver`, called:

| Sensor Type | Name |
|:----:|:----:|
| [`OrientationSensors`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IOrientationSensors.html)| `head_imu_acc_1x1` |
| [`ThreeAxisGyroscopes`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IThreeAxisGyroscopes.html) | `head_imu_acc_1x1` |
| [`ThreeAxisLinearAccelerometers`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IThreeAxisLinearAccelerometers.html)  |  `head_imu_acc_1x1` |
| [`ThreeAxisMagnetometers`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IThreeAxisMagnetometers.html) | `head_imu_acc_1x1` |

### Real Robot `ergoCubSN000`

On the real robot, instead the different sensors that are part of the head IMU all have different names, see https://github.com/robotology/robots-configuration/blob/18cfd97db62da019092cfa11133aa8667c1271ca/ergoCubSN000/hardware/inertials/head-inertial.xml#L33 :

| Sensor Type | Name |
|:----:|:----:|
| [`OrientationSensors`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IOrientationSensors.html)| `rfeimu_eul` |
| [`ThreeAxisGyroscopes`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IThreeAxisGyroscopes.html) | `rfeimu_gyro` |
| [`ThreeAxisLinearAccelerometers`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IThreeAxisLinearAccelerometers.html)  |  `rfeimu_acc` |
| [`ThreeAxisMagnetometers`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IThreeAxisMagnetometers.html) | `rfeimu_acc` |

### Discussion

Ideally the name of the simulated/model and real robot sensors should coincide, to permit to simplify writing that works both on the real robot and on the simulated robot, and permit to exploit the knoledge of positions of a robot coming from the model when writing software that runs on the real robot.

HosameldinMohamed, 2023-01-23:
CC @mebbaid

Nicogene, 2023-01-24:
A side question, are we sure that the `embObjIMU` publishes respect to this frame?
https://github.com/robotology/robots-configuration/blob/18cfd97db62da019092cfa11133aa8667c1271ca/ergoCubSN000/hardware/inertials/head-inertial.xml#L33

I am afraid that is a similar issue of https://github.com/robotology/icub-main/issues/812

traversaro, 2023-01-24:
> A side question, are we sure that the `embObjIMU` publishes respect to this frame? https://github.com/robotology/robots-configuration/blob/18cfd97db62da019092cfa11133aa8667c1271ca/ergoCubSN000/hardware/inertials/head-inertial.xml#L33
> 
> I am afraid that is a similar issue of [robotology/icub-main#812](https://github.com/robotology/icub-main/issues/812)

It all boils down of how do you call the frame where the `head_imu_acc_1x1` sensor is placed.

Nicogene, 2023-01-26:
Right now the frame name on the real robot is the one specified in `id`

- https://github.com/robotology/robots-configuration/blob/18cfd97db62da019092cfa11133aa8667c1271ca/ergoCubSN000/hardware/inertials/head-inertial.xml#L33
- https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjIMU/eo_imu_privData.cpp#L139-L140

If I am not mistaken all the measurements(gyro, magn, ACC, eul) are referred to the same reference frame, then using `id` is not good for us, we should add `framename` parameter(like @triccyx did https://github.com/robotology/icub-main/pull/821) and put the same for all the sensors, am I right?

traversaro, 2023-01-26:
To keep things simple can't we just change id to match the framename?

Nicogene, 2023-01-26:
I am afraid that that `id` is used by the firmware and has to be unique, we cannot assign the same for all the sensors.
@marcoaccame knows it for sure

traversaro, 2023-01-26:
If I remember correctly the `id` parameter was just introduced to implement the `getSensorName` and `getFrameName` methods, so I do not know if it is used in any other way. 
Furthermore, the MAS interfaces were kind of designed to have multiple "sensors" with the same name, as they were designed to permit to easily expose a single physical sensor such as an IMU as multiple "logical sensors" of different types, such as orientation, gyroscope, accelerometer, magnetometer.

However, even if we change the framename of the real robot, the sensorname would still be different between the simulated and real robot, so I am not sure it helps in any way for this issue.

Nicogene, 2023-01-27:
Chatting with @marcoaccame T2T he confirmed that those strings (field `id`) have to be unique because they are used for enabling the sensors, see https://github.com/robotology/robots-configuration/blob/18cfd97db62da019092cfa11133aa8667c1271ca/ergoCubSN000/hardware/inertials/head-inertial.xml#L47.

At this point I think the only solution here is to add `framename` parameter as has been done by @triccyx for the FT sensor, then this task from just fixing xml/yml requires some changes in the firmware.

Maybe in the meanwhile, we can agree on the frame names?

I would use this convention

`<part>_imu_<nr>`

maybe `<nr>` has to be 1-indexed in order to be compliant to these ids:

https://github.com/robotology/robots-configuration/blob/18cfd97db62da019092cfa11133aa8667c1271ca/ergoCubSN000/hardware/inertials/left_arm-eb24-j4_7-inertials.xml#L35

Nicogene, 2023-01-27:
> > At this point I think the only solution here is to add the `framename` parameter as has been done by @triccyx for the FT sensor, then this task from just fixing xml/yml requires some changes in the firmware.
> 
> Just to be aligned: how do you think that adding a `framename` could fix the inconsistency with the `get<type>SensorName()` call?

Adding `framename` would allow having the same return for `get<type>FrameName()` for all the logical sensors but we will have

 `get<type>SensorName()` ! =  `get<type>FrameName()`.
 
Actually, you are right, it does not fix this specific issue, `get<type>SensorName()` will remain different for each logic sensor but in gazebo we cannot split `libgazebo_yarp_imu` into separate logical sensors and name them as are named on the real robot, am I right?

`framename` will fix the visualization of the data on rviz2

traversaro, 2023-01-27:
> Maybe in the meanwhile, we can agree on the frame names?
>
> I would use this convention
>
> `<part>_imu_<nr>`
>
> maybe `<nr>` has to be 1-indexed in order to be compliant to these ids: [...]

Great! 

Two comments:
* perhaps we could use `<linkname>_imu_<nr>` instead of `<part>_imu_<nr>`? 
* probably we would need a label similar to `imu` for sensors with only gyro and  accelerometer? Or perhaps we could call all of them `imu`, but some of them would only have gyro and accelerometer

traversaro, 2023-01-27:
> Actually, you are right, it does not fix this specific issue, `get<type>SensorName()` will remain different for each logic sensor but in gazebo we cannot split `libgazebo_yarp_imu` into separate logical sensors and name them as are named on the real robot, am I right?

I guess that with  the following changes:
* add some configuration files complexity in gazebo-yarp-plugins
* modify the code of `gazebo-yarp-plugins` modifying the URDF files of iCub
* finding a new way to express sensors in URDF (as currently we use Gazebo's sensors, that do not include `accelerometer`, `gyroscope`, `magnetometer` as separate sensors but just `imu`
* modify the URDF models of iCub to use this new names, different for each type
this could be possible. Anyhow, I do not know if this work is worth it.

Perhaps a simpler solution is to modify the YARP device `embObjIMU` to read a new parameter (that could be `highlevelsensorname`, or just `sensorname`, as anyhow it is different from the `id` used by the firmware) and use that parameter as the return value of both `get<type>SensorName` and `get<type>FrameName`. To cover the use cases described in https://github.com/robotology/icub-main/issues/812 (that anyhow, it is not something we need on icub/ergocub models) we could even have a `framename`, but for the sake of trying to keep the configuration file complexity low I would make the `framename` optional and would just use `sensorname` if `framename` is not specified.

See also my comment in https://github.com/robotology/icub-main/issues/812#issuecomment-1242710127 .

> At this point I think the only solution here is to add framename parameter as has been done by @triccyx for the FT sensor, then this task from just fixing xml/yml requires some changes in the firmware.

Which changes are necessary in the firmware? I tought we just need to modify the YARP device code in icub-main, right? See https://github.com/robotology/icub-main/pull/822/files that did a similar change.

traversaro, 2023-01-27:
Anyhow @Nicogene if you want we can quickly align in person/Teams.

Nicogene, 2023-01-27:
> Which changes are necessary in the firmware? I tought we just need to modify the YARP device code in icub-main, right? See [robotology/icub-main#822 (files)](https://github.com/robotology/icub-main/pull/822/files) that did a similar change.

Exactly, we need to do similar changes, I noticed that exists 2 types of imu devices in `icub-main`

- https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjIMU/embObjIMU.h
- https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjInertials/embObjInertials.h

(They correspond to these two "entity" types https://github.com/robotology/icub-firmware-shared/blob/09ab8b4b14639adcf61ec420d131e9f89de417eb/eth/embobj/plus/comm-v2/icub/EoAnalogSensors.h#L99-L100 ?)

> Anyhow @Nicogene if you want we can quickly align in person/Teams.

I think since this task became bigger than expect it is good to have an alignment with all the actors for starting off on the right foot. 
There are several points that have to be touched on and changed we have to be sure that we all agree before starting.
Maybe a meeting @pattacini @traversaro @marcoaccame and me?

pattacini, 2023-01-27:
>Maybe a meeting @pattacini @traversaro @marcoaccame and me?

Yep!
Let me see if we can arrange for 1h this afternoon, otherwise we will schedule it next week.

Nicogene, 2023-01-27:
> * [robotology/icub-main@`master`/src/libraries/icubmod/embObjIMU/embObjIMU.h](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjIMU/embObjIMU.h?rgh-link-date=2023-01-27T11%3A04%3A25Z)
> 
> * [robotology/icub-main@`master`/src/libraries/icubmod/embObjInertials/embObjInertials.h](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/embObjInertials/embObjInertials.h?rgh-link-date=2023-01-27T11%3A04%3A25Z)

After the T2T meeting, we realized that it is worth now focusing on `embObjIMU` since `embObjInertials` is the old device and they seem to cover the same functionalities.

>Perhaps a simpler solution is to modify the YARP device embObjIMU to read a new parameter (that could be highlevelsensorname, or just sensorname, as anyhow it is different from the id used by the firmware) and use that parameter as the return value of both get<type>SensorName and get<type>FrameName

We could start with this implementation and then if need something like https://github.com/robotology/icub-main/issues/812 can be addressed later.

commit dee4a756 (2023-01-31): Change the sensorname of the IMU

It fixes #36
closed 2023-01-31 by Nicogene