mesh-iit/ergocub-software #37: FK/IK with ergoCub

pattacini, 2023-01-25:
This is more a sort of discussion about how to implement forward and inverse kinematics on ergoCub.

With @traversaro we thought that this time we can start off with using `idyntree` from the ground up, by initializing the kinematic chains straight from the urdf.

In this respect, what kinds of IK algorithms are available from `idyntree`, considering also the important aspects of joint limits avoidance, tasks prioritization, and the like?

Also, `ikin`-based components could be still employed by feeding the DH parameters from files (once yielded via `idyntree`). In fact, we are not going to publish hand-made kinematic chains based on DH convention in this repo.

CC'ing people in the know: @xenvre @mfussi66 @Nicogene @traversaro @lrapetti

traversaro, 2023-01-25:
Probably also interesting for @DanielePucci @S-Dafarra @GiulioRomualdi .

GiulioRomualdi, 2023-01-25:
A possibility is also to use the [ik implemented in blf](https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/src/IK). It implements a modular based ik structured as a QP problem. 

This is the software we are currently using in the walking pipeline.

pattacini, 2023-01-26:
Even if `ikin` gets enables somehow through files, as reported above, one important thing to bear in mind is that `iKinGazeCtrl` cannot be used with ergoCub as the head is completely different, missing out the eyes, which are dealt with tightly alongside the neck in iCub.

One possibility could be adapting/extending the R1 gaze control, which is merely a ports-based module, meaning there is no YARP device driver and no C++ API. In this sense, `idyntree` can still replace `ikin`.

traversaro, 2023-01-26:
So, I will try to recap the available IKs:

| Class   | Technique  | Joint limits avoidance? |  Task prioritization |  General Notes |  
|:--------:|:---------:|:------------------------:|:-------------------:|:---------------:|
| `iCub::iKin::iKinIpOptMin` | Position-based IK using ipopt to solve the nonlinear optimization problem on kinematic chains | Yes | Yes | To load URDF, you need to first load a URDF model via `iDynTree::ModelLoader` and then convert it to an `iCub::iKin::iKinLimb` via `iDynTree::iKinLimbFromModel`. It is only deals with kinematic chains. |
| `iDynTree::InverseKinematics` | Position-based IK using ipopt to solve the nonlinear optimization problem on kinematic chains | Yes | Yes, either via using IK tasks as cost terms with different weights, or by using some IK tasks as constraint in the nlopt problem, and some other as cost terms to minimize. | The implementation is quite complex and did not received a lot of love in the latest years, no symbolic hessian are implemented. |
| `BipedalLocomotion::IK::QPInverseKinematics` | Integration-based IK in which a series of QP are solved at the velocity level, and the result is then integrated. | ??? |  Yes, in the stack of task sense | |

traversaro, 2023-01-26:
@GiulioRomualdi I am not sure about "Joint limits avoidance?" in BipedalLocomotion::IK::QPInverseKinematics .

DanielePucci, 2023-01-26:
Probably another feature that we may add to the table above is the computational time of the IK iteration on a given laptop/computer - I know that this would probably require some reference problem and work, so it's ok if we do not add it now. I just wanted to highlight the point since the computational time should differ substantially - at least in theory and given some of our experience [1,2] - when solving position-based nonlinear problems  and integration-based QP SOT. Often another feature that differs substantially is the *smoothness* of the solution, although this is more a qualitative judgement that we seldom evaluated quantitatively 

[1]: https://www.mdpi.com/1999-4893/13/10/266/htm
[2]: https://www.youtube.com/watch?v=arKB7M8ccpo

GiulioRomualdi, 2023-01-26:
> @GiulioRomualdi I am not sure about "Joint limits avoidance?" in BipedalLocomotion::IK::QPInverseKinematics .

Not yet implemented but if needed I can do it

xEnVrE, 2023-01-26:
Just a clarification about two comments I read above:

> Also, ikin-based components could be still employed by feeding the DH parameters from files (once yielded via idyntree).

> load a URDF model via iDynTree::ModelLoader and then convert it to an iCub::iKin::iKinLimb via iDynTree::iKinLimbFromModel

Are these alternatives or both required to use an `iCub::iKin`-based solution?

traversaro, 2023-01-26:
> Just a clarification about two comments I read above:
> 
> > Also, ikin-based components could be still employed by feeding the DH parameters from files (once yielded via idyntree).
> 
> > load a URDF model via iDynTree::ModelLoader and then convert it to an iCub::iKin::iKinLimb via iDynTree::iKinLimbFromModel
> 
> Are these steps alternatives or both required to use an `iCub::iKin`-based solution?

They are more and less the same thing. In what @pattacini suggested, you convert the chain in the URDF file to a `.ini` file via the `urdf2dh` command provided by https://github.com/robotology/idyntree-yarp-tools#urdf2dh , and then you load the `.ini` created file via the `iKinLimb::fromLinksProperties` method. In this way, you avoid to have to link iDynTree in your application/library, and you can continue to just link iKin. Otherwise (and this is what I was describing) you can do the same steps in the code, without going through the intermediate .ini file. But the two approaches are basically equivalent from the  point of view of the final result.

pattacini, 2023-01-26:
Hi @GiulioRomualdi 

> Not yet implemented but if needed I can do it

And how do you make sure that you won't bump into the HW limits?

S-Dafarra, 2023-01-26:
> Hi @GiulioRomualdi
> 
> > Not yet implemented but if needed I can do it
> 
> And how do you make sure that you won't bump into the HW limits?

One approach could be the one presented in Sec. III.B of https://arxiv.org/pdf/1909.07669.pdf

pattacini, 2023-01-26:
> One approach could be the one presented in Sec. III.B of https://arxiv.org/pdf/1909.07669.pdf

Ok, so there exists the joint limit avoidance, implemented in the space of joint velocities.

Is this correct or is it not implemented on the robot yet?

S-Dafarra, 2023-01-26:
> > One approach could be the one presented in Sec. III.B of https://arxiv.org/pdf/1909.07669.pdf
> 
> Ok, so there exists the joint limit avoidance, implemented in the space of joint velocities.
> 
> Is this correct or is it not implemented on the robot yet?

It has been implemented for the human side, not for the robot side. We are working on making the two converge into a unique implementation.

GiulioRomualdi, 2023-02-10:
> > @GiulioRomualdi I am not sure about "Joint limits avoidance?" in BipedalLocomotion::IK::QPInverseKinematics .
> 
> Not yet implemented but if needed I can do it

I implemented the task in this PR: https://github.com/ami-iit/bipedal-locomotion-framework/pull/603

pattacini, 2023-04-28:
I deem we can close this issue.
Feel free to open it up again if required.

closed 2023-04-28 by pattacini