mesh-iit/ergocub-software #38: Problem when installing the repo

gabrielecaddeo, 2023-01-26:
Hi everybody, I am trying to use ergoCub on simulation. I firstly installed the repo in a local directory that I created but when I try to insert the model on Gazebo, the log gives me this output:
`
[Msg] Waiting for model database update to complete...
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_root_link_prt.stl]
[Err] [Visual.cc:3030] No mesh specified
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_l_hip_1_prt.stl]
`
The environment variables are set correctly.
It seems like it can find the urdf but it cannot find the associated stl files (the same error is repeated for every stl in the urdf).  However, if I select as install prefix the install folder of the superbuild, I can insert ergoCub without any problem. Is this normal? Thanks

traversaro, 2023-01-26:
I think this is the problem I discussed in the third bullet point in https://github.com/robotology/robotology-superbuild/pull/1331#issuecomment-1403400759, basically you also need to add `<install-prefix>/share` to the `GAZEBO_MODEL_PATH`. You do not experience this when using the robotology-superbuild because in that case that folder is already in the `GAZEBO_MODEL_PATH`.

Nicogene, 2023-01-26:
> I think this is the problem I discussed in the third bullet point in https://github.com/robotology/robotology-superbuild/pull/1331#issuecomment-1403400759, basically you also need to add [...]

Hi @gabrielecaddeo I added this to the README, please let us know if it solves your problem.

closed 2023-01-26 by Nicogene

gabrielecaddeo, 2023-01-26:
Yes guys, problem solved, thanks!

commit 3e27fbf8 (2023-01-26): Update README.md

Fixes #38