mesh-iit/ergocub-software #39: Make ergoCub ROS2 compatible

pattacini, 2023-01-26:
We're required to make ergoCub talk ROS2 as we did for iCubGenova11.
We will contribute to this task together w/ @randaz81 and colleagues.

I've put this task on our first PI14 sprint, which is Feb 6 – Feb 17.

cc @traversaro

randaz81, 2023-02-01:
FYI @SimoneMic

randaz81, 2023-02-01:
After a quick chat with @traversaro:
- we have Ubuntu 22.04 on the main CPU board (the one which controls the robot joints). This is ok.
- we have Ubuntu 20.04 on the Nvidia board (the one which controls the lidar and the realsense). This is an issue that needs to be addressed, since we need to publish these data on ROS2, and Ubuntu 22.04 is required. Possible solutions (never tested) include:
    - use of conda
    - use of docker

traversaro, 2023-02-01:
fyi @GiulioRomualdi

Nicogene, 2023-02-02:
In this PR (https://github.com/robotology/robots-configuration/pull/429) I added the xml files for running ROS2 nws, for now in the remapper I put also the hands in order to not add other xmls (`remapper_no_hands`), but for this open discussion (https://github.com/robotology/community/discussions/609) we will not be able to use the hands in ROS2 (visualize/control).

Nicogene, 2023-02-08:
Today I installed:

- `ros-humble-desktop` on `ergocub-laptop` via apt
- `ros-humble-ros-base` on `ergocub-torso` via apt

I tested successfully https://github.com/robotology/robots-configuration/pull/429 (Incredible, it is the first time I write correctly the remapper XML :D).

I can echo all the topics (`/joint_state`, `/l_arm_ft` etc).

I didn't work on `ergocub-head`, we should decide together how to install ros-humble there.

maggia80, 2023-02-08:
browsing around, I found this webpage: https://news.accelerationrobotics.com/ros-2-humble-in-nvidia-xavier-agx-with-yocto/

Nicogene, 2023-02-08:
> browsing around, I found this webpage: [news.accelerationrobotics.com/ros-2-humble-in-nvidia-xavier-agx-with-yocto](https://news.accelerationrobotics.com/ros-2-humble-in-nvidia-xavier-agx-with-yocto/)

It seems wild, but we can try on the xavier we wave spare 👍🏻

traversaro, 2023-02-08:
> > browsing around, I found this webpage: [news.accelerationrobotics.com/ros-2-humble-in-nvidia-xavier-agx-with-yocto](https://news.accelerationrobotics.com/ros-2-humble-in-nvidia-xavier-agx-with-yocto/)
> 
> It seems wild, but we can try on the xavier we have spare 👍🏻

Note that in that case you would also need to install YARP and its dependencies via Yocto, or build the superbuild on the top of the yocto packages. Personally I will give it at least a try via conda-forge/robostack before trying Yocto.