mesh-iit/ergocub-software #42: Fix up the order of the fingers joints in YARP

pattacini, 2023-02-02:
We spotted that the thumb and index abductions (handled by a dedicate board) come at the end of the joint set. This goes a bit against the order we have on iCub and, at least, segregates the two joints of the thumb apart.

We need to harmonize the sequence both in the real robot and in the simulated one.

Nicogene, 2023-02-07:
Right now the joints' order for the hands in gazebo is:
```
l_thumb_add l_thumb_prox l_thumb_dist l_index_add l_index_prox l_index_dist l_middle_prox l_middle_dist l_ring_prox l_ring_dist l_pinkie_prox l_pinkie_dist
```
I don't know if this is the desired order, but we should have the same joint order on the simulated and real robot.

First of all, I noticed that the names on the real robot do not match the names in simulation that follows [this table](https://icub-tech-iit.github.io/documentation/hands/hands_mk5/):

See https://github.com/robotology/robots-configuration/blob/b31e860ece0e0ad539a6f2d74b19cf1f4cf60931/ergoCubSN000/hardware/mechanicals/right_arm-eb24-j11_12-mec.xml#L12.

`r_hand_thumbmetacarpus` -> `r_thumb_add` ?
`r_hand_indexadduction` -> `r_index_add` ?

pattacini, 2023-02-07:
>l_thumb_add l_thumb_prox l_thumb_dist l_index_add l_index_prox l_index_dist l_middle_prox l_middle_dist l_ring_prox l_ring_dist l_pinkie_prox l_pinkie_dist

This list refers to the physical joints but what we control is instead:

>l_thumb_add l_thumb_oc l_index_add l_index_oc l_middle_oc l_ring_pinky_oc

I took the freedom to use the suffix `oc` (standing for open-close) since we cannot talk about either proximal or distal joints in this case. We can even think of removing the suffix `oc`, if it's not self-explaining.

I deem that the order you reported, @Nicogene, is ok and best resembles what we have on iCub.
Actually, on iCub `l_index_add ` should come as the first on, but in that case the adduction includes also the ring and the pinky, which is not our case with ergoCub. After all, the order looks fine to me.

>r_hand_thumbmetacarpus -> r_thumb_add ?
>r_hand_indexadduction -> r_index_add ?

This is correct 👍🏻

pattacini, 2023-02-07:
cc @fiorisi @marcoaccame @davidetome @maggia80

pattacini, 2023-02-07:
For me it's ok; let's wait for others to chime in and agree on this.
cc @fiorisi @traversaro 

Another point is related to the pinky. It's not just pinky but rather `ring_pinky`.
I'm in favor of the latter convention, which looks more explicit to me.

traversaro, 2023-02-07:
I agree with the `_oc` suffix and with `ring_pinky`, thanks for handling this!

Nicogene, 2023-02-08:
For having this joint list:

> l_thumb_add l_thumb_oc l_index_add l_index_oc l_middle_oc l_ring_pinky_oc

I tried to configure the remapper like this

```xml
<device xmlns:xi="http://www.w3.org/2001/XInclude" name="left_arm-mc_remapper" type="controlboardremapper">
    <paramlist name="networks">
        <elem name="left_arm_joints1">(  0  1 0 1 )  </elem>
        <elem name="left_arm_joints2">(  2  3 0 1 )  </elem>
        <elem name="left_arm_joints3">(  4  6 0 2 )  </elem>
        <elem name="left_arm_joints5">(  7  7 0 0 ) </elem>
        <elem name="left_arm_joints4">(  8  8 0 0 ) </elem>
        <elem name="left_arm_joints5">(  9  9 1 1 ) </elem>
        <elem name="left_arm_joints4">(  10 12 1 3 ) </elem>
    </paramlist>
    <param name="joints"> 13 </param>
    <action phase="startup" level="5" type="attach">
        <paramlist name="networks">
            <elem name="left_arm_joints1"> left_arm-eb2-j0_1-mc   </elem>
            <elem name="left_arm_joints2"> left_arm-eb4-j2_3-mc   </elem>
            <elem name="left_arm_joints3"> left_arm-eb31-j4_6-mc  </elem>
            <elem name="left_arm_joints4"> left_arm-eb23-j7_10-mc </elem>
            <elem name="left_arm_joints5"> left_arm-eb25-j11_12-mc </elem>
        </paramlist>
    </action>
    <action phase="shutdown" level="20" type="detach" />
</device>
```

But the `yarprobotinterface` parser complains about duplication of parameters (e.g. `left_arm_joints5`).
Is there an alternative to the configuration I wrote?

Otherwise, if we put `thum_add` and `index_add` at the beginning of the list we should not have issues.

cc @traversaro @davidetome

traversaro, 2023-02-08:
> But the `yarprobotinterface` parser complains about duplication of parameters (e.g. `left_arm_joints5`).
> Is there an alternative to the configuration I wrote?

I would use the `axesNames` way of configuring the controlboardremapper instead of the networks way, see https://yarp.it/latest/classControlBoardRemapper.html#details . As a side effect I personally find `axesNames`  more mantainable and readable, but that may be just my taste.

Nicogene, 2023-02-08:
> I would use the axesNames way of configuring the controlboardremapper instead of the networks way, see [yarp.it/latest/classControlBoardRemapper.html#details](https://yarp.it/latest/classControlBoardRemapper.html#details) . As a side effect I personally find axesNames more mantainable and readable, but that may be just my taste.

I never used `axesNames` for remapping, do you have an example of configuration file?

In the meanwhile I found a workaround to attach multiple times the same device:
https://github.com/robotology/robots-configuration/pull/435

I tested on the robot and we have the names/order we want, it is a little bit dirty but it works

pattacini, 2023-02-08:
>I never used axesNames for remapping, do you have an example of configuration file?

Ok 👍🏻 
I'll wait for any feedback on this before merging.

traversaro, 2023-02-08:
> I never used `axesNames` for remapping, do you have an example of configuration file?

It is not in the controlboardremapper case, but the syntax is the same for all devices that use this parameter: https://github.com/robotology/robots-configuration/blob/b31e860ece0e0ad539a6f2d74b19cf1f4cf60931/iCubMoscow01/estimators/wholebodydynamics.xml#L6 . You just to list the name of the axis you want to consider in the order you want to consider them.

Nicogene, 2023-02-09:
I committed here https://github.com/Nicogene/robots-configuration/tree/fix/ergocubHands2 the version proposed by @traversaro, I will test it on the robot.

pattacini, 2023-02-09:
> I committed here https://github.com/Nicogene/robots-configuration/tree/fix/ergocubHands2 the version proposed by @traversaro, I will test it on the robot.

It worked!
Job merged via:
- https://github.com/robotology/robots-configuration/pull/436

pattacini, 2023-02-09:
@Nicogene, I would continue here the work to update the Gazebo side as well.
Later, we can harmonize the naming convention also on the FW with a dedicated issue.

closed 2023-02-09 by Nicogene

pattacini, 2023-02-09:
>Later, we can harmonize the naming convention also on the FW with a dedicated issue.

Closing in favor of #56.

PR #54 (2023-02-09): urdf: change the names of the joints of the hands in order to be compliant with the real robot
See https://github.com/robotology/robots-configuration/pull/436

It fixes #42