mesh-iit/ergocub-software #43: Add up the FT sensors to the hips

pattacini, 2023-02-04:
As per the title.

pattacini, 2023-02-04:
@Nicogene is dealing with this in https://github.com/robotology/robots-configuration/tree/fix/addHipFtErgocub.

pattacini, 2023-02-04:
Hi @Nicogene 

I've rebased again the fix branch on top of the latest devel.
Please, do a pull rebase on your local system to get aligned.

Nicogene, 2023-02-08:
Today I tried the xml files on `ergoCub` and I noticed that `yarprobotinterface` does not start because seems that the FT sensors in the hips are not connected, I cannot find them with the `FirmwareUpdater`, but they are there, they are visible.

cc @AntonioConsilvio @sgiraz

pattacini, 2023-02-08:
cc @Fabrizio69 too.

pattacini, 2023-02-08:
>Hi, we noticed that there aren't FT sensors in the hips. So we need replace the dummy (the fake FT sensor mounted on the robot) to two real sensors.

We don't have 2 brand new FT58; the only 2 FT58 we have are on iCub3.

pattacini, 2023-02-08:
cc @giulioromualdi @traversaro

lrapetti, 2023-02-08:
> > Hi, we noticed that there aren't FT sensors in the hips. So we need replace the dummy (the fake FT sensor mounted on the robot) to two real sensors.
> 
> We don't have 2 brand new FT58; the only 2 FT58 we have are on iCub3.

I don't know what would be the timing for having new FT58. The only option to start working with wbd, is to remove the FT from the model.

pattacini, 2023-02-08:
>I don't know what would be the timing for having new FT58.

It'll take some time.

>@Nicogene @pattacini would it be possible for you to apply this patch?

Yep 👍🏻

Nicogene, 2023-02-08:
> > I don't know what would be the timing for having new FT58. The only option to start working with wbd, is to remove the FT from the model. @Nicogene @pattacini would it be possible for you to apply this patch?
> 
> Just mentioning that this change is needed only on the `ergoCubSN000` model urdf, and not the `ergoCubGazeboV1` which has `wholebodydynamics` device running as of this morning.
> 
> If the plan is to change the simulation model as well to keep the consistency, we may also change the configurations to match the change on the real robot before opening the relevant PR @HosameldinMohamed

For the time being I removed them only from the `ergocubSN000` see https://github.com/icub-tech-iit/ergocub-software/pull/45

closed 2023-02-08 by pattacini