mesh-iit/ergocub-software #47: Left shoulder yaw and camera tilt not coherent between robot and model

lrapetti, 2023-02-08:
Today @GiulioRomualdi and @mebbaid spotted that there is a mismatch in the direction of the shoulder yaw joint and the camera tilt between the model and the robot. 

This was spotted comparing the motion in Gazebo and the real robot when increasing the value of the joint from `yarpmotorgui`.

Here for example is the comparison for the shoulder:

https://user-images.githubusercontent.com/35487806/217552978-82aedd43-ec9c-40c4-9bbd-cacecf55c1bd.mp4

All the other joints have been checked and are coherent.

cc @pattacini @Nicogene @sgiraz

pattacini, 2023-02-08:
Thanks heaps for reporting!

pattacini, 2023-02-08:
Hi @mfussi66,

We need first to understand what is the intended correct direction of the rotation and then fix either the robot XML file or the URDF.

mfussi66, 2023-02-09:
> We need first to understand what is the intended correct direction of the rotation and then fix either the robot XML file or the URDF.

By checking within Gazebo, that the URDF needs changing. The camera should tilt upwards with a positive angle, to be consistent with the head pitch.

pattacini, 2023-02-09:
> to be consistent with the head pitch.

True.

Sorry for being nitpicking but what is the direction of movement of the iCub eyes with respect to the head pitch? Just to understand our "heritage".

mfussi66, 2023-02-09:
> Sorry for being nitpicking but what is the direction of movement of the iCub eyes with respect to the head pitch? Just to understand our "heritage".

In Gazebo iCub3 moves neck pitch and eyes tilt in opposite directions, see:

[image]

Not sure if it's the same on the robot.

mfussi66, 2023-02-09:
Regarding the shoulder yaw, I think that also needs a fix at the URDF level, since the iCub3 Gazebo model is consistent with the real ergoCub (assuming Gazebo iCub3 and real iCub3 are consistent between each other).

pattacini, 2023-02-09:
> > Sorry for being nitpicking but what is the direction of movement of the iCub eyes with respect to the head pitch? Just to understand our "heritage".
> 
> In Gazebo iCub3 moves neck pitch and eyes tilt in opposite directions, see:
> Not sure if it's the same on the robot.

Thanks for checking 👍🏻 
The physical iCub 2.x is aligned with iCub in Gazebo as there's a number of software working on both the domains correctly.

Just an example:
- https://github.com/vvv-school/assignment_make-it-roll

This doesn't mean that we need to copy out the same directions on ergoCub though.

closed 2023-02-09 by Nicogene