mesh-iit/ergocub-software #48: Upload yarpmotorgui configurations

lrapetti, 2023-02-08:
Hi, in order to use the robots we generally rely on some `yarpmotorgui` home configurations.

In the past we had some discussions on where to store those files, but we ended up storing them in https://github.com/robotology/whole-body-controllers/tree/master/utilities/homePositions even if they might not be related to the controllers stored in that repo. 

Since we (me @GiulioRomualdi and @mebbaid) are now preparing some new files for the ergoCub robot (actually we have prepared a configuration for the robot standing on two feet [homePositions.ini.txt](https://github.com/icub-tech-iit/ergocub-software/files/10687711/homePositions.ini.txt)), we were wondering if it would be the case to find a better location for those files (e.g. inside `ergocub-software`).

cc @pattacini @traversaro @gabrielenava

lrapetti, 2023-02-08:
One possibility could be to have a folder `ergocub-software/app/homePositions/robots/...`. 

Then we should also understand where those files would be installed. (in `.../install/share/ergoCub/robots/${ROBOT_NAME}/` together with the model?)

pattacini, 2023-02-08:
Hi @lrapetti 
Hosting those files in this repo is certainly possible 👍🏻 
However, it is also required to put the cmake machinery to install them.

Another possibility is to host the files in a dedicated subfolder of the ergoCubSN000 folder inside `robots-configuration`, which I personally tend to like more.

lrapetti, 2023-02-09:
> Another possibility is to host the files in a dedicated subfolder of the ergoCubSN000 folder inside `robots-configuration` (and fiddle just a little bit with the already available cmake file), which I personally tend to like more.

That seems reasonable also for me. I don't remember in the past if this option was considered, and in case why it was discarded. Then we will probably upload the file in https://github.com/ami-iit/component_ergocub/issues/197#issuecomment-1423050620 together with the other updates.

pattacini, 2023-02-10:
Related PR available at:
- https://github.com/robotology/robots-configuration/pull/437.

lrapetti, 2023-02-10:
PR has been merged, we can close it!

closed 2023-02-10 by lrapetti