mesh-iit/ergocub-software #49: Face expression stopped to work

lrapetti, 2023-02-09:
In order to enable wbd on `ergocub-torso`, we have installed the robotology-superbuild dependencies using the [`install_apt_dependencies.sh`](https://github.com/robotology/robotology-superbuild/blob/master/scripts/install_apt_dependencies.sh) script. While doing this, however, the following message was raised:
[image]
we continued, but after a while we realized the face expressions stopped to work (I guess it was the service that was supposed to be restarted). After restarting the robot motors and CPU, the face expression seems to be still off:
[image]

cc @GiulioRomualdi @mebbaid

lrapetti, 2023-02-09:
cc @Nicogene @pattacini 

Let us know if this is the right place to open this kind of issues, or if you prefer them to be opened in https://github.com/robotology/icub-tech-support/issues/

Nicogene, 2023-02-09:
By the way the oled screen is attached to ergocub-head, it may be possible that the screen stopped working independently what you did on ergocub-torso

pattacini, 2023-02-09:
>At the same time, we can open an issue here for what regards the development of a SW layer to deal with the OLED.

See https://github.com/icub-tech-iit/ergocub-software/issues/50.

Closing then.

closed 2023-02-09 by pattacini