mesh-iit/ergocub-software #50: Develop proper facial expressions handler

pattacini, 2023-02-09:
See https://github.com/icub-tech-iit/ergocub-software/issues/49#issuecomment-1423881680.

martinaxgloria, 2023-03-10:
In the last few days I started implementing the RFModule of the `emotionHandler` and yesterday I had a look at the technologies that can be used to show images. In particular, I will try to include `OpenCV` modules to read an image and open it in a dedicated window when a specific rpc command is sent. 

Moreover, @mfussi66 suggested that a possible alternative could be the use of `yarpview`, but I haven't explored it yet.

I'm going to try both to understand which is the more suitable solution.

pattacini, 2023-03-10:
Regarding `yarpview`, it will be definitely a constraint when/if we will be displaying videos as well.
So, I wouldn't use it.

martinaxgloria, 2023-03-10:
Ok, I will test OpenCV then.

pattacini, 2023-03-10:
> Ok, I will test OpenCV then.

Just to clarify, OpenCV and yarpview are not mutually exclusive technologies as the former is a library for handling images in C++, whereas the latter is a yarp based app for displaying images traveling on the yarp network.

martinaxgloria, 2023-03-14:
I implemented in this personal [fork](https://github.com/martinaxgloria/ergocub-software/tree/feat/emotionHandler_module) a first draft of the RFModule that uses the rpc protocol. In particular, I tested the OpenCV technology for displaying an expression (i.e. png/jpeg images) when an rpc command is received on my PC and it seems to work so far (even if I found only really low-quality images without copyright).

I don't know if we could consider this technology as the final one, but I think this could be an interesting starting point.

cc @Nicogene @pattacini

Nicogene, 2023-03-14:
> I don't know if we could consider this technology as the final one, but I think this could be an interesting starting point.

Absolutely, you can open a PR displaying your module at work, and this is material for a demo during the review 👍🏻

martinaxgloria, 2023-03-14:
Perfect, thank you!

pattacini, 2023-03-14:
Probably, we can even meet up earlier than Friday and see the demo live and then continue with what's next.

martinaxgloria, 2023-03-14:
It's ok for me 👍

pattacini, 2023-03-14:
I've got some time tomorrow afternoon.

closed 2023-03-15 by Nicogene