mesh-iit/ergocub-software #53: Make ergoCub ROS2 compatible – Stint 2

pattacini, 2023-02-09:
Following up on #39, we're required to mount ROS2 on the head and test the ROS2 functionalities on the whole robot.

Nicogene, 2023-02-20:
📆  ergocub booked for tomorrow (21st February) 16.00-17.00

cc @martinaxgloria

Nicogene, 2023-02-20:
Alongside @martinaxgloria, we managed to install `ros humble` via conda packages on the NVIDIA Xavier Agx we have on bench.

We followed these steps:

- [Install mamba-forge](https://github.com/robotology/robotology-superbuild/blob/master/doc/install-mambaforge.md#linux)
- [Create `ros_humble` env and install ros-humble-desktop](https://github.com/RoboStack/ros-humble#installation) (skipping the openGL part).
- `mamba install ros-humble-test-msgs`

We compiled `yarp-devices-ros2` flawlessly activating `ros_humble` environment.

Tomorrow @martinaxgloria will repeat the same steps on the Xavier of the robot (`ergocub-head`).

traversaro, 2023-02-20:
> We compiled `yarp-devices-ros2` flawlessly activating `ros_humble` environment.

I guess you installed YARP as well from source, right?

> skipping the openGL part

Probably you do not need them if you do not compile anything that uses qt, but if you need them (for example to compile YARP GUIs?) you can install them via:
~~~
mamba install mesa-libgl-devel-cos7-aarch64 mesa-dri-drivers-cos7-aarch64 libselinux-cos7-aarch64 libxdamage-cos7-aarch64 libxxf86vm-cos7-aarch64 libxext-cos7-aarch64 xorg-libxfixes
~~~ 

See https://github.com/icub-tech-iit/ergocub-software/issues/53#issuecomment-1437044515 .

Nicogene, 2023-02-20:
> I guess you installed YARP as well from source, right? 

Yes! We installed the superbuild some time ago ( before the mission at Rome ).

traversaro, 2023-02-20:
> > I guess you installed YARP as well from source, right?
> 
> Yes! We installed the superbuild some time ago ( before the mission at Rome ).

In general apt and conda packages are not ABI-compatible, so if you are mixing in the same binary (`yarp-devices-ros2`) libraries compiled with apt dependencies (such as YARP) and libraries compiled with conda-forge/robostack dependencies (ROS humble) you could get runtime crashes. In a nutshell, you can't mix libraries from apt and conda-forge/robostack, or even between different environenments.

martinaxgloria, 2023-02-21:
Today, supervised by @Nicogene, I successfully installed `ros_humble` on the Xavier of the robot (`ergocub_head`) by following the steps listed above.

pattacini, 2023-02-21:
Superb!

martinaxgloria, 2023-02-22:
> In general apt and conda packages are not ABI-compatible, so if you are mixing in the same binary (`yarp-devices-ros2`) libraries compiled with apt dependencies (such as YARP) and libraries compiled [...]

Since we've installed `ros_humble` in a `conda environment`, all the packages are stored into a single folder easily removable if those problems will occur. Moreover, the possible solution is to switch to a docker environment.

traversaro, 2023-02-22:
Just to be aligned, another solution is just to use/compile YARP using conda-forge dependencies.

Nicogene, 2023-02-23:
After a T2T alignment(@randaz81, @pattacini, @traversaro, @martinaxgloria) we considered the possibility of using docker but it has the following cons:

1. Our "official" docker image `superbuild-ros2` is out of order (https://github.com/icub-tech-iit/docker-deployment-images/issues/1) and moreover, we should have an image also with cuda, then probably have to build an ad-hoc image for the Xaview
2. Also if we manage to have this image, there is the possibility that we cannot get the HW acceleration (GPU) in the docker container (Since Xavier uses a lot of patched packets from the jetpack).

Unfortunately also recompiling the `robotology-superbuild` or other repositories inside the conda environment could lead to point 2.

Since we have not a silver bullet, the idea, for now, is to keep it as it is, try to run the ROS2 nws of the rgbd sensors and see if with rqt we can read the images from the realsense.

In case of problems we can think about recompiling the repositories under the conda environment and see if it fixes the problem.

martinaxgloria, 2023-02-24:
Today, @Nicogene and I had a meeting in which we discussed about the solutions proposed in the yesterday alignment. 
The first thing we did after activating the conda environment ros_humble, we tried to simply load a ros2 yarp plugin: `yarp plugin wrenchStamped_nws_ros2`. Unfortunately, as @traversaro predicted, we obtain the following error:
```
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so:
  no error
/lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so)
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so:
  no error
/lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so)
[ERROR] |yarp.os.YarpPluginSettings| Error while opening /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so:
  no error
/lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so)
[ERROR]     Cannot find or load shared library
```

Then, we tried to install the `robotology-superbuild` inside the conda environment, but before that we need to install the OpenGL conda package:
```
mamba install mesa-libgl-devel-cos7-aarch64 mesa-dri-drivers-cos7-aarch64 libselinux-cos7-aarch64 libxdamage-cos7-aarch64 libxxf86vm-cos7-aarch64 libxext-cos7-aarch64 xorg-libxfixes
```
and `ace` and `ipopt`
```
conda install -c conda-forge ace ipopt
```
After recompiling also `yarp-device-realsense2` and `yarp-devices-ros2`, we finally got yarp ros2 plugin working:

```
Yes, this is a YARP plugin
  * library:        /usr/local/src/robot/robotology-superbuild/build/install/lib/yarp/yarp_wrenchStamped_nws_ros2.so
  * system version: 5
  * class name:     WrenchStamped_nws_ros2
  * base class:     yarp::dev::DeviceDriver
```

We tried the rgbd nws ros2 with the realsense launching the yarprobotinterface with this XML:
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE robot PUBLIC "-//YARP//DTD yarprobotinterface 3.0//EN" "http://www.yarp.it/DTD/yarprobotinterfaceV3.0.dtd">

 <devices>

    <device xmlns:xi="http://www.w3.org/2001/XInclude" name="realsenseD450" type="realsense2">
        <group name="SETTINGS">
            <param name="depthResolution"> (640 480) </param>
            <param name="rgbResolution">  (640 480) </param>
            <param name="framerate">    30  </param>
            <param name="enableEmitter"> true </param>
            <param name="alignmentFrame"> RGB </param>
        </group>
        <group name="HW_DESCRIPTION">
            <param name="clipPlanes"> (0.2 10.0) </param>
        </group>
        <group name="QUANT_PARAM">
            <param name="depth_quant"> 2 </param>
        </group>
        <param name="rotateImage180"> false </param>
    </device>

    <device xmlns:xi="http://www.w3.org/2001/XInclude" name="RGBDWrapperyarp" type="rgbdSensor_nws_yarp">
        <param name="period"> 0.03 </param>
        <param name="name">    /ergocub/depthCamera </param>
        <action phase="startup" level="5" type="attach">
            <paramlist name="networks">
                <elem name="subdevicergbd"> realsenseD450 </elem>
            </paramlist>
        </action>
        <action phase="shutdown" level="5" type="detach" />
    </device>
    <device xmlns:xi="http://www.w3.org/2001/XInclude" name="RGBDWrapperros2" type="rgbdSensor_nws_ros2">       
        <param name="period"> 0.033 </param>       
       <param name="color_topic_name">/camera_rgbd/color/image_rect_color</param>       
        <param name="depth_topic_name">/camera_rgbd/depth/image_rect</param>       
        <param name="color_frame_id">depth_center</param>       
        <param name="depth_frame_id">depth_center</param>       
        <param name="node_name">rgbdsensor_ros_node</param>       
        <action phase="startup" level="5" type="attach">           
        <paramlist name="networks">               
        <elem name="subdevicergbd"> realsenseD450 </elem>           
        </paramlist>       
        </action>       
        <action phase="shutdown" level="5" type="detach" />   
        </device>
 </devices>
```
The yarprobotinterface started smoothly. With the ros2 topic list we saw the topics relative to the realsense opened and with ros2 topic echo we managed to see data coming from the camera. 

Next step: we will do all the steps listed above on the real robot.

martinaxgloria, 2023-02-27:
📆 ergocub booked for March 1st from 10:30 to 12:30.

Nicogene, 2023-03-01:
Chatting with @lrapetti and @xenvre, since ROS2 is not used for the demos of 6-7th March, we decided to put on hold this activity and take it back after the demos, in order to reduce the uncertainties.

traversaro, 2023-03-02:
Just for alignment, now on conda-forge we have librealsense packages for arm64 and compiled with cuda support (see https://github.com/conda-forge/librealsense-feedstock/pull/28 and https://github.com/conda-forge/librealsense-feedstock/pull/32). I do not know if that is enough to liobtain the same result you have now with the custom build, but in case you want to try to install realsense in a conda environment with cuda support it should be sufficient to run:
~~~
mamba install librealsense=*=*cuda
~~~

Nicogene, 2023-03-06:
> Just for alignment, now on conda-forge we have librealsense packages for arm64 and compiled with cuda support (see [conda-forge/librealsense-feedstock#28](https://github.com/conda-forge/librealsense-feedstock/pull/28), [conda-forge/librealsense-feedstock#19](https://github.com/conda-forge/librealsense-feedstock/issues/19) and [conda-forge/librealsense-feedstock#32](https://github.com/conda-forge/librealsense-feedstock/pull/32)). I do not know if that is enough to liobtain the same result you have now with the custom build, but in case you want to try to install realsense in a conda environment with cuda support it should be sufficient to run:
> 
> ```
> mamba install librealsense=*=*cuda
> ```

This is maybe something we can try on our xavier we have spare, I would keep the installation as we tested the previous week.

I have booked `ergoCub` for tomorrow morning installing ROS2 on `ergocub-head`

cc @martinaxgloria

martinaxgloria, 2023-03-07:
Today I successfully installed the entire `ros_humble environment` on `ergocub-head` and, at the end, launching `rqt` we were able to see the images coming from the realsense as you can see in the figure below.

[image]

Moreover, the conda environment is already activated and ready to be used when you connect to the ergocub-head, so you can skip the `conda activate ros_humble` command.

cc @Nicogene @traversaro @pattacini

traversaro, 2023-03-07:
It may be worth to mention this on the ergoCub chat, this is the first time we have a conda-forge based robotology-superbuild installed on a robot system, so it could be unexpected for some users.

martinaxgloria, 2023-03-07:
Here the rate at which the images are published read from the topic `/camera_rgbd/color/image_rect_color`

[image]

and `/camera_rgbd/depth/image_rect`

[image]

This values are related to uncompressed images which are read from Wi-Fi.

Nicogene, 2023-03-07:
Also checking from `rqt` I would say that the result we achieved is good for closing this activity, if needed we can proceed with a `stint 3` later.

PS: after upgrading to YARP 3.8.0 we managed also to compile `yarp-devices-ros2` to the latest master

cc @traversaro @pattacini @randaz81

closed 2023-03-07 by Nicogene