mesh-iit/ergocub-software #56: Fix up the order of the fingers joints in FW

pattacini, 2023-02-09:
This is a reminder for us to clean up the FW level as well, following up on https://github.com/icub-tech-iit/ergocub-software/issues/42#issuecomment-1424130079.

sgiraz, 2023-02-20:
For sure we would have to change the names within `eObrd_portpos_t` in `icub-firmware-shared`:

```cpp
typedef enum
{
    eobrd_portpos_hand_thumb            = 0,
    eobrd_portpos_hand_index            = 1,
    eobrd_portpos_hand_medium           = 2,
    eobrd_portpos_hand_pinky            = 3,
    eobrd_portpos_hand_thumbmetacarpus  = 4,
    eobrd_portpos_hand_thumbrotation    = 5,
    eobrd_portpos_hand_indexadduction   = 6,
    
    eobrd_portpos_none                  = 31,    // as ... eobrd_port_none
    eobrd_portpos_unknown               = 30     // as ... eobrd_port_unknown
} eObrd_portpos_t;
```

pattacini, 2023-02-22:
Worked started in:
- https://github.com/robotology/icub-firmware/pull/346

pattacini, 2023-02-23:
Putting in review while waiting for the tests to be done.

marcoaccame, 2023-02-27:
> Worked started in:
> 
> * [HandMK5: fix names accordingly to the new joint names robotology/icub-firmware#346](https://github.com/robotology/icub-firmware/pull/346)
> * [HandMK5: fix names accordingly to the new joint names robotology/robots-configuration#444](https://github.com/robotology/robots-configuration/pull/444)
> * [HandMK5: fix names accordingly to the new joint names  robotology/icub-firmware-shared#78](https://github.com/robotology/icub-firmware-shared/pull/78)

Hi @Nicogene,  I think that we can close that in a short time. 

Pls rebase your forks vs robotology/devel and I will have a look at `icub-firmware` and `icub-firmware-shared`.

marcoaccame, 2023-02-27:
OK @Nicogene, the PRs are ready. I have added another one to force the use of the new icub-firmware-shared:
- https://github.com/robotology/icub-main/pull/859

Instead we don't need any change in the binaries (see comment in [here](https://github.com/robotology/icub-firmware/pull/346#issuecomment-1446183508)).

Morever, I dont think that we need tests of that, so I will begin the merging of the PRs in a short while.

marcoaccame, 2023-02-27:
> Morever, I dont think that we need tests of that, so I will begin the merging of the PRs in a short while (I will however just run the parsing of the xml files in icub-main).

Tests OK. I will start merging

marcoaccame, 2023-02-27:
@Nicogene: two PRs are missing (icub-main + robots-configuration) and then it is OK to close the issue

Nicogene, 2023-02-27:
All the PRs have been merged, thanks a lot @marcoaccame!
Closing.

closed 2023-02-27 by Nicogene