mesh-iit/ergocub-software #59: Having a ROS2 pkg for RViz meshes

SimoneMic, 2023-02-22:
Hello,

I was wondering if it would be possible to have an automatic process that outputs a ROS2 pakage for the ergocub URDF and meshes. To be used in the `robot_state_publisher` and `RViz`  pkgs in ROS2.

The actual model fails to load the meshes on RViz, while it's fine for the `robot_state_publisher` and the `TF2` pakage.

The error message on Rviz:
```
Errors loading geometries:
• for link 'chest':
Could not load mesh resource 'package://ergoCub/meshes/simmechanics/sim_ecub_chest_prt.stl
```
[image]

TFs frames: displayed correctly
[image]

I was wondering if there's something already existing for R1, or if anyone has already incurred in this issue and has some tips on how to solve this.
It's mostly a visual thing, so it's not of the maximum importance, but just to point out for discussion.

Thanks!

lrapetti, 2023-02-22:
cc @Nicogene

traversaro, 2023-02-22:
This can be done by:
* Installing a well-formed `package.xml` file in `<install_prefix>/share/ergoCub`
* Install an empty file named `ergoCub` in `<install_prefix>/share/ament_index/resource_index/packages` (see https://github.com/robotology/icub-models/blob/master/CMakeLists.txt#L181)
* Add the `<install_prefix>` path to `AMENT_PREFIX_PATH` env variable, unless it is not already there (this is the case for the `robotology-superbuild` install prefix if you sourced the `setup.sh` script)

See https://github.com/robotology/icub-models/issues/177 and https://github.com/robotology/icub-models/issues/178 for a related issue and PR.

Nicogene, 2023-02-23:
Hi @SimoneMic, if you want you can implement what @traversaro suggests and open a PR, otherwise I will do it, but I am not sure when I can handle this

SimoneMic, 2023-02-28:
Hi
Sorry for the late reply, I have implemented the solutions on my fork for the demo.
Afterwards we can merge it.

Thanks.

traversaro, 2023-02-28:
Great! In the meanwhile it is convenient to have at least the pointer of the releant branches, commit, even if there is no PR. I guess that the branch is https://github.com/SimoneMic/ergocub-software/tree/navigation and the relevant commit are https://github.com/SimoneMic/ergocub-software/commit/4fc2f14809e467565bfe1b2092cd5d6af9ee2408 and https://github.com/SimoneMic/ergocub-software/commit/1baa1a9051d6f3656cbb5837e8ce7969f4f6ee0c, right?

SimoneMic, 2023-02-28:
> Great! In the meanwhile it is convenient to have at least the pointer of the releant branches, commit, even if there is no PR. I guess that the branch is https://github.com/SimoneMic/ergocub-software/tree/navigation and the relevant commit are [SimoneMic@4fc2f14](https://github.com/SimoneMic/ergocub-software/commit/4fc2f14809e467565bfe1b2092cd5d6af9ee2408) and [SimoneMic@1baa1a9](https://github.com/SimoneMic/ergocub-software/commit/1baa1a9051d6f3656cbb5837e8ce7969f4f6ee0c), right?

Yes, exactly

SimoneMic, 2023-03-17:
Closing since discussed in #83

closed 2023-03-17 by SimoneMic