mesh-iit/ergocub-software #60: ergoCubGazeboV1 - joints l_shoulder_yaw and l_wrist_yaw joints go in hardware fault

xEnVrE, 2023-02-27:
While testing the simulation using the `ergoCubGazeboV1` model, I noticed that often the joints `l_shoulder_yaw` and `l_wrist_yaw` go in hardware fault, especially the `l_elbow` joint.

By playing a bit with the tuning of the PID in simulation, I noticed that the problem seems to disappear by removing the derivative action for joints `l_shoulder_yaw` and `l_wrist_yaw` - not saying this is the cause, maybe the derivative action is just emphasizing the effect caused by the real problem, e.g. wrongly scaled inertias, etc..

Is it ok to open a PR in order to change them in the file used by the urdf generator, i.e. https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/simmechanics/data/ergocub/conf/gazebo_ergocub_left_arm.ini? Otherwise, at the moment, the left arm is quite not usable.

Are the current values of the PID been inherited from iCub2.x simulated models or have they been tuned somehow?

Thanks

Nicogene, 2023-02-28:
Hi @xEnVrE!

> Is it ok to open a PR in order to change them in the file used by the urdf generator, i.e. [master/urdf/simmechanics/data/ergocub/conf/gazebo_ergocub_left_arm.ini](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/simmechanics/data/ergocub/conf/gazebo_ergocub_left_arm.ini?rgh-link-date=2023-02-27T17%3A57%3A47Z)? Otherwise, at the moment, the left arm is quite not usable.

Sure! For the time being we can fix this issue like you said.
Is this happening only on the left arm?
Because I would do the same change also on the right arm.

xEnVrE, 2023-02-28:
Thanks @Nicogene

lrapetti, 2023-02-28:
Hi @Nicogene, I have updated and installed the latest `master`, but the issue is still occurring.

When launching this gazebo world: [ergoCub_lifting_40.zip](https://github.com/icub-tech-iit/ergocub-software/files/10849327/ergoCub_lifting_40.zip)

I immediately get failure of left wrist and shoulder yaw:
```
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  2  torque too big! (  -2423.91  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  4  torque too big! (  -2251.08  )
```

lrapetti, 2023-02-28:
> Hi @Nicogene, I have updated and installed the latest `master`, but the issue is still occurring.
>
> When launching this gazebo world: [...]

Removing the derivative gains as mentioned in https://github.com/icub-tech-iit/ergocub-software/issues/60#issue-1601675795 solves the issue also for this case.

@Nicogene let me know if you want me to open a new issue or re-open this one

xEnVrE, 2023-02-28:
I tested it again and for me the issue seems solved now without the need to disable the derivative action.

cc @Nicogene @lrapetti

lrapetti, 2023-03-01:
> I tested it again and for me the issue seems solved now without the need to disable the derivative action.

I will double check again if the corrected model was loaded in my test

PR #61 (2023-02-28): urdf: fix small inertia issues on left arm
This issue was due to a leftover of old link name of the forearm

It fixes #60