mesh-iit/ergocub-software #74: Check and eventually fix ergoCub lower leg frames orientation

paolo-viceconte, 2023-03-14:
By visualizing the ergoCub link frames in Gazebo I noticed that the ergoCub link frames are all oriented as iCubV3 but the lower leg ones, which are instead rotated of 180 degs around the x axis (z pointing downwards rather than upwards):

[image]

If this is not intended for specific purposes, maybe it is worth to have them oriented as all the other links in the chain.

Nicogene, 2023-09-20:
Hi @Mick3Lozzo,

I noticed that you committed the fixes after pulling `cad-mechanics`, next time please drop a comment here in order I receive the notification.

Your changes align the knee frames but here is the resulting urdf

[image]

Also checking the diff seems that the whole part has been roll-ed up for having the z up

```diff
-    <origin xyz="0.01060000963260986 0.007899987075259199 -0.17970000000000103" rpy="-3.1415926535897274 -8.69101748565412e-14 -1.2193157698906137e-6"/>
-    <axis xyz="-2.0458175022504648e-17 1 2.5382740632965355e-17"/>
+    <origin xyz="0.01060000963260985 0.007899987075259289 -0.17970000000000058" rpy="6.892643261935339e-14 -6.650159361880639e-14 -1.2193157700904425e-6"/>^M
+    <axis xyz="1.7937066152151813e-16 -1 3.2454535731621664e-15"/>
```

Probably something went wrong in the regeneration of the simp rep? 

It has to be investigated

cc @pattacini @mfussi66

pattacini, 2023-09-29:
@Mick3Lozzo 

Can you please resume here the current status?

Mick3Lozzo, 2023-09-29:
@pattacini I corrected and updated both the CAD and shrink-wrapped models (for the `ergoCub_1-1 `sim model).
The shrink-wrapped model does not use the CSYSes as a constraint, so the legs' placement is correct.

[image]

[This problem](https://github.com/icub-tech-iit/ergocub-software/issues/74#issuecomment-1727786937) still has to be investigated.

Nicogene, 2023-10-04:
Today I managed to understand what was the cause of this problem.
Basically, I forgot to update also the meshes after the change of orientation of the csys. Now it looks fine:

[image]

PR:
- https://github.com/icub-tech-iit/ergocub-software/pull/181

> [!WARNING]  
> In order to fix this issue it is needed to change the simulation model and reassemble it. Since ergoCubV1 cad has not been changed/fixed for a while, this fix will be carried only on ergoCubV1_1 models.

cc @Mick3Lozzo @pattacini

PR #181 (2023-10-04): ergoCubV1_1: Fix reference frame in the lower legs
It fixes #74. 
> [!WARNING]  
> In order to fix this issue it is needed to change the simulation model and reassemble it. Since ergoCubV1 cad has not been changed/fixed for a while, this fix will be carried only on ergoCubV1_1 models.
commit 308728ca (2023-10-04): ergoCubV1_1: Fix reference frame in the lower legs

It fixes #74. Note that ergoCubV1 model will be not fixed since the the simulation model has not been changed since a while
closed 2023-10-04 by Nicogene