mesh-iit/ergocub-software #75: Add white diffuse color to the robot links for proper visualization in Ignition Gazebo simulator

paolo-viceconte, 2023-03-14:
When trying to visualize the ergoCub model in Ignition Gazebo, the robot is visualized all in white:

[image]

This is instead the result if a white diffuse color is applied to all the robot links:

[image]

The lines added to the urdf model to get the above can be found [here](https://github.com/paolo-viceconte/paper_viceconte_2021_ral_adherent/blob/4b250e5e519324ea6ec8ea720885268574672678/src/adherent/model/ergoCubGazeboV1_xsens/ergoCubGazeboV1_xsens.urdf#L3410-L3871).

This issue is to track the possibility of including these changes in the ergoCub urdf, since they can be of interest for future development in Ignition Gazebo.

cc @traversaro @S-Dafarra @GiulioRomualdi

Nicogene, 2023-03-14:
I don't think it is possible to add it right now, I check `simmechanics-to-urdf` but there is not any option that handles visual, @traversaro correct me if I am wrong.
I would open an issue there for adding this feature, even if we are working on creating our own creo2urdf converter

paolo-viceconte, 2023-03-16:
Closing this for the time being, since it could be just related to a specific development setup. We will reopen it if it appears to be a more general issue.

closed 2023-03-16 by paolo-viceconte