mesh-iit/ergocub-software #77: Develop proper facial expressions handler – Stint 2

Nicogene, 2023-03-15:
### Task description
This is the follow-up of #50.

We ought to address these points:

- [ ] Refactor the module in order to use [thrift](https://yarp.it/latest/idl_thrift.html)
- [ ] Investigate and overcome the `waitkey` issue
- [ ] investigate and overcome the fullscreen issue
- [ ] When the module starts a default emotion or a black image has to be shown
- [ ] Manage the images using the [context/ResourceFinder machinery](https://yarp.it/latest/yarp_resource_finder_tutorials.html)

### Definition of done

All those points are addressed by a PR and it has been merged

cc @pattacini

martinaxgloria, 2023-03-16:
Yes, I already tried to show simple mp4 videos and it's possible to do 👍

martinaxgloria, 2023-03-16:
Regarding the `waitKey()` issue, today I found a way to trigger the function and get the reply before the loop starts by using the function `pollKey()` in the HighGUI module. 

The video below shows how it works now. In this way, it's possible to switch from one image to another without remaining blocked. Of course it could be improved but this was the biggest limit of the module so far.

https://user-images.githubusercontent.com/114698424/225690455-5b42960c-596c-4e1c-86ac-912c286f3b0f.mp4

pattacini, 2023-03-18:
Regarding this:

>Manage the images using the [context/ResourceFinder machinery](https://yarp.it/latest/yarp_resource_finder_tutorials.html)

I propose organizing the conf files as follows:

```
ergoCubEmotions/
├── config.ini
└── expressions
    ├── images
    │   ├── exp_img_0.png
    │   └── exp_img_1.png
    │   └── ...
    └── videos
        ├── exp_mov_0.mp4
        └── exp_mov_1.mp4
        └── ...
```

The content of the main file `config.ini` containing the configuration options may look like this:

```ini
[general]
num_expressions 10

[expression_0]
name happy
type image
file images/exp_img_0.png

[expression_1]
name sad
type image
file images/exp_img_1.png

...

[expression_9]
name cunny
type video
file videos/exp_mov_0.mp4
```

The module may then manage a `std::map` to handle the association between the tag of type string (e.g., `happy`, `sad`...) and the corresponding information for which we could create an ad-hoc structure to contain the type, the file, and the object[^1] to display (image or video).

[^1]: By exploiting the object-oriented paradigm of C++, we can design a class to wrap images and videos and expose a simple API for displaying purposes.

martinaxgloria, 2023-03-20:
Hi @pattacini, thank you for the advice. Do you suggest handling this task before thrifting the module or it's the same?

pattacini, 2023-03-20:
> Hi @pattacini, thank you for the advice. Do you suggest handling this task before thrifting the module or it's the same?

They're two independent sub-tasks. You could complete thrifting the module first and then address its configuration. While doing the latter, you may also need to adjust thrift methods.

martinaxgloria, 2023-03-20:
Clear, thank you!

martinaxgloria, 2023-03-24:
Today I implemented the context/ResourceFinder mechanism to manage the images and the update module is [here](https://github.com/martinaxgloria/ergocub-software/tree/feat/stint2). If you're ok, I could open a PR to see the current state of the implementation.

cc @Nicogene @pattacini

pattacini, 2023-03-24:
Fine with me!

pattacini, 2023-03-27:
PR available here:
- https://github.com/icub-tech-iit/ergocub-software/issues/87

closed 2023-03-29 by Nicogene