mesh-iit/ergocub-software #78: WBD segfaults in simulation

G-Cervettini, 2023-03-15:
Hi all, when I try to run WBD in a clean 2023.02 distro with ergocub-software v0.2.0 I got the following error.

```
YARP_CLOCK=/clock yarprobotinterface --config conf/launch_wholebodydynamics_ecub.xml
[INFO] |yarp.os.Port|/IITICUBLAP264/yarprobotinterface/150602/clock:i| Port /IITICUBLAP264/yarprobotinterface/150602/clock:i active at tcp://10.240.2.10:10098/
[INFO] |yarp.os.Network| Success: port-to-port persistent connection added.
[INFO] |yarp.os.Time| Waiting for clock server to start broadcasting data ...
[INFO] |yarp.os.impl.PortCoreInputUnit|/IITICUBLAP264/yarprobotinterface/150602/clock:i| Receiving input from /clock to /IITICUBLAP264/yarprobotinterface/150602/clock:i using tcp
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
DEBUG] Reading file /home/guglielmo/robotology-superbuild/build/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[WARNING] Invalid syntax while loading /home/guglielmo/robotology-superbuild/build/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml at line 1 . Expected document of type robot . Found devices
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /home/guglielmo/robotology-superbuild/build/install/share/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[INFO] Yarprobotinterface was started using the following enable_tags: 
[INFO] Yarprobotinterface was started using the following disable_tags: 
[DEBUG] List of all enable attributes found in the include tags: 
[DEBUG] List of all disable attributes found in the include tags: 
[DEBUG] Preprocessor complete in:  0 s
[WARNING] Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
[WARNING] *************************************************************************************
* yarprobotinterface 'portprefix' parameter does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name    *
*     name:    full port prefix name with leading '/',  e.g.  /robotName      *
* A temporary automatic fix will be done for you, but please fix your config file   *
*************************************************************************************
[INFO] |yarp.os.Port|/ergoCubGazeboV1/yarprobotinterface| Port /ergoCubGazeboV1/yarprobotinterface active at tcp://10.240.2.10:10051/
[INFO] startup phase starting...
[INFO] Opening device torso_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/torso"), ("local" = "/wholeBodyDynamics/torso")]
[DEBUG] |yarp.dev.PolyDriver|torso_mc| Parameters are (device remote_controlboard) (id torso_mc) (local "/wholeBodyDynamics/torso") (remote "/ergocubSim/torso") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/rpc:o| Port /wholeBodyDynamics/torso/rpc:o active at tcp://10.240.2.10:10052/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/command:o| Port /wholeBodyDynamics/torso/command:o active at tcp://10.240.2.10:10053/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/stateExt:i| Port /wholeBodyDynamics/torso/stateExt:i active at tcp://10.240.2.10:10054/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/rpc:o| Sending output from /wholeBodyDynamics/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/command:o| Sending output from /wholeBodyDynamics/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|torso_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device left_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/left_arm"), ("local" = "/wholeBodyDynamics/left_arm")]
[DEBUG] |yarp.dev.PolyDriver|left_arm_mc| Parameters are (device remote_controlboard) (id left_arm_mc) (local "/wholeBodyDynamics/left_arm") (remote "/ergocubSim/left_arm") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/rpc:o| Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://10.240.2.10:10055/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/command:o| Port /wholeBodyDynamics/left_arm/command:o active at tcp://10.240.2.10:10056/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/stateExt:i| Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://10.240.2.10:10057/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/rpc:o| Sending output from /wholeBodyDynamics/left_arm/rpc:o to /ergocubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/command:o| Sending output from /wholeBodyDynamics/left_arm/command:o to /ergocubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/left_arm/stateExt:i| Receiving input from /ergocubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|left_arm_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device right_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/right_arm"), ("local" = "/wholeBodyDynamics/right_arm")]
[DEBUG] |yarp.dev.PolyDriver|right_arm_mc| Parameters are (device remote_controlboard) (id right_arm_mc) (local "/wholeBodyDynamics/right_arm") (remote "/ergocubSim/right_arm") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/right_arm/rpc:o| Port /wholeBodyDynamics/right_arm/rpc:o active at tcp://10.240.2.10:10058/
[INFO] |yarp.os.Port|/wholeBodyDynamics/right_arm/command:o| Port /wholeBodyDynamics/right_arm/command:o active at tcp://10.240.2.10:10059/
[INFO] |yarp.os.Port|/wholeBodyDynamics/right_arm/stateExt:i| Port /wholeBodyDynamics/right_arm/stateExt:i active at tcp://10.240.2.10:10060/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/right_arm/rpc:o| Sending output from /wholeBodyDynamics/right_arm/rpc:o to /ergocubSim/right_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/right_arm/command:o| Sending output from /wholeBodyDynamics/right_arm/command:o to /ergocubSim/right_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/right_arm/stateExt:i| Receiving input from /ergocubSim/right_arm/stateExt:o to /wholeBodyDynamics/right_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|right_arm_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device left_leg_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/left_leg"), ("local" = "/wholeBodyDynamics/left_leg")]
[DEBUG] |yarp.dev.PolyDriver|left_leg_mc| Parameters are (device remote_controlboard) (id left_leg_mc) (local "/wholeBodyDynamics/left_leg") (remote "/ergocubSim/left_leg") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_leg/rpc:o| Port /wholeBodyDynamics/left_leg/rpc:o active at tcp://10.240.2.10:10061/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_leg/command:o| Port /wholeBodyDynamics/left_leg/command:o active at tcp://10.240.2.10:10062/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_leg/stateExt:i| Port /wholeBodyDynamics/left_leg/stateExt:i active at tcp://10.240.2.10:10063/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_leg/rpc:o| Sending output from /wholeBodyDynamics/left_leg/rpc:o to /ergocubSim/left_leg/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_leg/command:o| Sending output from /wholeBodyDynamics/left_leg/command:o to /ergocubSim/left_leg/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/left_leg/stateExt:i| Receiving input from /ergocubSim/left_leg/stateExt:o to /wholeBodyDynamics/left_leg/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|left_leg_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device right_leg_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/right_leg"), ("local" = "/wholeBodyDynamics/right_leg")]
[DEBUG] |yarp.dev.PolyDriver|right_leg_mc| Parameters are (device remote_controlboard) (id right_leg_mc) (local "/wholeBodyDynamics/right_leg") (remote "/ergocubSim/right_leg") (robotName ergoCubGazeboV1)
[... cut ...]
[DEBUG] wholeBodyDynamics Statistics: Attaching all IMUs took  0 s.
[DEBUG] wholeBodyDynamics Statistics: Calibrating offsets.
[DEBUG] wholeBodyDynamics Statistics: Calibrating took  0 s.
[DEBUG] wholeBodyDynamics Statistics: Starting
[INFO] All actions for action level 15 of startup phase started. Waiting for unfinished actions.
[INFO] All actions for action level 15 of startup phase finished.
[INFO] startup phase finished.
[INFO] run phase starting...
[DEBUG] yarprobotinterface running happily
Segmentation fault (core dumped)
```

traversaro, 2023-03-16:
@G-Cervettini  Can you run `yarprobotinterface --config conf/launch_wholebodydynamics_ecub.xml` under valgrind or a debugger (for example gdb), to get a backtrace of the failure?

G-Cervettini, 2023-03-22:
I guess we can close this as completed for now.

closed 2023-03-22 by G-Cervettini