mesh-iit/ergocub-software #79: Missing "kinematic" links of the fingers skin

xEnVrE, 2023-03-17:
At the moment, the following links representing the fingertip skin are available:

`r_hand_thumb_skin_0`, ..., `r_hand_thumb_skin_11`
`r_hand_index_skin_0`, ..., `r_hand_index_skin_11`
`r_hand_middle_skin_0`, ..., `r_hand_middle_skin_11`
`r_hand_ring_skin_0`, ..., `r_hand_ring_skin_11`

`l_hand_thumb_skin_0`, ..., `l_hand_thumb_skin_11`
`l_hand_index_skin_0`, ..., `l_hand_index_skin_11`
`l_hand_middle_skin_0`, ..., `l_hand_middle_skin_11`
`l_hand_ring_skin_0`, ..., `l_hand_ring_skin_11`

Should not we have also additional links for the left/right little - or pinkie - fingers?

cc @Nicogene

Nicogene, 2023-03-17:
> Should not we have also additional links for the left/right little - or pinkie - fingers?

They should be there

https://github.com/icub-tech-iit/ergocub-software/blob/caf4c5d4b3318a7f4b92caa2f9ce4957cf124a1f/urdf/simmechanics/data/ergocub/ERGOCUB_all_options.yaml#L477-L513

https://github.com/icub-tech-iit/ergocub-software/blob/caf4c5d4b3318a7f4b92caa2f9ce4957cf124a1f/urdf/simmechanics/data/ergocub/ERGOCUB_all_options.yaml#L663-L698

They are not in the generated urdf?

Nicogene, 2023-03-17:
Ah! It doesn't :(

https://github.com/search?q=repo%3Aicub-tech-iit%2Fergocub-software+l_hand_pinkie_skin_3&type=code 

But because there is a typo in the csys

https://github.com/icub-tech-iit/ergocub-software/blob/caf4c5d4b3318a7f4b92caa2f9ce4957cf124a1f/urdf/simmechanics/data/ergocub/SIM_ERGOCUB_1-0.xml#L3354-L3354

xEnVrE, 2023-03-17:
> They are not in the generated urdf?

Unless I am mistaken, I cannot find them in the urdf right now.

Nicogene, 2023-03-17:
Fix delivered in #80

xEnVrE, 2023-03-17:
Thanks @Nicogene

PR #80 (2023-03-17): urdf: fixed taxels exportations for pinkies
It fixes #79