mesh-iit/ergocub-software #82: Weird behavior with the simulated finger mechanisms in Gazebo

xEnVrE, 2023-03-17:
When moving the fingers in simulation I noticed that there are some part of the mechanism - probably even not required for the scope of the simulation - that move strangely:

[image]

https://user-images.githubusercontent.com/6014499/225933133-10316d8a-91a0-4a37-9cca-3e93a7d2c8ed.mp4

I though it was useful to notify that in case it reveals any kind of issue within the urdf generation process.

cc @mfussi66 @Nicogene

pattacini, 2023-03-17:
>Because it is wanted. 😃

Can you elaborate on this @Lawproto?

xEnVrE, 2023-03-17:
> what do you mean by "strangely"?

As I said in the comment, the simulation of that lever is not probably required as the constraint between the two phalanges is enforced software-wise, [here](https://github.com/robotology/gazebo-yarp-plugins/blob/4b35f025c93aa51870c57e81d97fbcee273ed876/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L1136)  while Gazebo does not use the lever actually.

Nonetheless, I would say that the mesh should not contain that part probably. Not sure if that was wanted.

xEnVrE, 2023-03-17:
I just updated the title of the issue to better reflect the actual problem.

Lawproto, 2023-03-20:
> Nonetheless, I would say that the mesh should not contain that part probably. Not sure if that was wanted.

This is questionable, I have to integrate the lever in some link to match the mass of the CAD model. It was convenient for me to have it in the last phalanx. Since our CAD-to-URDF is convoluted enough, I would personally live with the aestethic imperfection.

> I just updated the title of the issue to better explain the actual problem.
It is not the goal of the URDF to capture exactly how the real-life mechanism operates. As long as the relation between the angles of the links (aka the phalanges) is correct while they rotate, I cannot see any problem.
@xEnVrE I am starting to miss the whole point of this issue, can we please discuss F2F?

pattacini, 2023-03-20:
>It is not the goal of the URDF to capture exactly how the real-life mechanism operates.

That's true outright 👍🏻 

However, having a mesh that can somehow interact with the environment during grasping or that does not represent the geometry of the hand could be a headache sometimes.

Perhaps, we can get away with less in the mid-term by removing the mesh and assigning its weight to the phalanx via software.

Lawproto, 2023-03-20:
> However, having a mesh that can somehow interact with the environment during grasping or that does not represent the geometry of the hand could be a headache sometimes.

Ok, I can see your point.

> Perhaps, we can get away with less in the mid-term by removing the mesh and assigning its weight to the phalanx via software.

Knowing all of this, I could directly integrate the same lever in the medial phalanx for the next URDF generation instead, so it does not lean out.

xEnVrE, 2023-03-20:
Me and @Lawproto discussed f2f on it and we agreed that the URDF cannot 100% reproduce the real mechanism and never will as it does not offer the formalism to do it.

We also agreed on the fact that if it was only an aesthetic issue we could probably neglect it. There might be however other connected issues - we did not consider before - such as unwanted collisions or difficulties in using tools like [idyntree-model-simplify-shapes](https://github.com/robotology/idyntree#idyntree-model-simplify-shapes) which constructs simplified representation of the URDF  and approximates each solid shape of the model with its axis aligned bounding box.

pattacini, 2023-03-22:
We agreed with @Nicogene and @mfussi66 to adjust manually the meshes of the phalanxes to remove the levers. The weights and the mass matrices will remain unvaried.

@Lawproto, let's remember that we should take this into account whenever in the future we yield the urdf of the hands once again.

pattacini, 2023-03-23:
Hi @xEnVrE 

We will attack this task in our next sprint: Apr 3 – Apr 14.

xEnVrE, 2023-03-23:
Below you can find the meshes before and after the manual removal of the levers:

### Before

[image]

### After

[image]

Of course the cut is "approximate" as the surfaces of the mesh are not that smooth.

If they are ok I can open a PR with the updated meshes (of course also the right hand will be included!).

cc @pattacini @Nicogene

Lawproto, 2023-03-24:
> Of course the cut is "approximate" as the surfaces of the mesh are not that smooth.

@xEnVrE it seems a great job to me!

pattacini, 2023-03-24:
Let's do it!
Thanks again @xEnVrE!

PR #85 (2023-03-24): Remove levers of the finger actuation mechanism from the meshes of the distal fingers
As per the title.

Meshes were manually adapted using the following `Meshlab` tools:
- "Select faces in a rectangular region" [image]
- "Close Holes" [image]

Fixes #82