mesh-iit/ergocub-software #88: Develop proper facial expressions handler – Stint 3 ❌ FIXME

Nicogene, 2023-03-29:
### Task description
TBD

### Definition of done
TBD

martinaxgloria, 2023-03-31:
About the first task of the list, I have just found out that OpenCV doesn't support `gif`, therefore if we want to move in that direction, there would be the necessity to rely on other libraries (I found for example [ImageMagick](https://imagemagick.org/)) which can handle also animated images). On the other hand, I can go ahead with .mp4 videos that can be displayed with OpenCV and try to use short videos that probably could be looped like gifs.

cc @Nicogene @pattacini

pattacini, 2023-03-31:
>About the first task of the list, I have just found out that OpenCV doesn't support gif

It may be convenient to post here the link to that resource for the record.

>On the other hand, I can go ahead with .mp4 videos that can be displayed with OpenCV and try to use short videos that probably could be looped like gifs.

Animated GIF were meant as a (temporary) workaround, so yes, I would still stick to OpenCV and videos in the first place.

pattacini, 2023-03-31:
Maybe, a GIF can be opened in OpenCV through the videocapture?

https://answers.opencv.org/question/226582/cv2imread-fail-to-open-gif-image/

martinaxgloria, 2023-03-31:
As said before, I tried with VideoCapture with and without some of those [flags](https://docs.opencv.org/4.x/d4/d15/group__videoio__flags__base.html#ga023786be1ee68a9105bf2e48c700294d) (like `CAP_FFMPEG`) but it doesn't work.

martinaxgloria, 2023-04-06:
After further tests, I found out that the Xavier doesn't work with OpenCV versions higher than `4.2.0`, but the function `pollKey()` which is currently used in ergoCubEmotions module was not already implemented in that version as you can see [here](https://docs.opencv.org/4.2.0/d7/dfc/group__highgui.html) in the official documentation of the High-level GUI module.

pattacini, 2023-04-06:
Great!

martinaxgloria, 2023-04-06:
Here the results on the OLED

[image]
[image]

Nicogene, 2023-04-06:
Today we had a brief "demo", and we showed to @maggia80 the progress on the development of the `ergocubEmotions` module, and the emotions as static images worked smoothly also on the OLED connected to the Xavier, very well done @martinaxgloria 🎖️ 

> Investigate how to build a "packet" to give to the designer for testing

We chatted about this point and it came out that in the end for the designer is important to see how it appears and not how works our module, then we thought to simply give them an OLED with a spare mask. In this way, they will easily see the impact of their work.

During the demo the part for displaying the video didn't work probably for the [same issue about the multithreading](https://github.com/icub-tech-iit/ergocub-software/issues/88#issuecomment-1499177868), then it is needed also a refactoring for that part

martinaxgloria, 2023-04-07:
I open a draft PR:

- https://github.com/icub-tech-iit/ergocub-software/pull/89

I implemented the part for displaying video and it works on my laptop, but I  have to do some tests on the Xavier in order to check if it works also on it.

cc  @Nicogene @pattacini

martinaxgloria, 2023-04-12:
This morning I've done some tests on the Xavier in order to display a simple video in loop and here is the result! So, now we are able to show both video and static images.

https://user-images.githubusercontent.com/114698424/231425196-2a16192f-1ee5-4643-b56a-457021dcc8aa.MOV

cc @Nicogene @pattacini @maggia80

martinaxgloria, 2023-04-12:
> I open a draft PR:
> 
> * [Develop proper facial expressions handler – Stint 3 #89](https://github.com/icub-tech-iit/ergocub-software/pull/89)

I moved the PR from draft to review.

closed 2023-04-12 by Nicogene