mesh-iit/ergocub-software #90: Where is the ergoCub startup application stored?

lrapetti, 2023-04-07:
Hi all, yesterday I was looking for the source file of the ergoCub startup application `ergoCubStartupV0.xml` that is installed in the ergoCub robot laptop in `/home/ergocub/.local/share/yarp/applications` but I was not able to find it.

The only file I have found is the one stored in `ergocub-software` (i.e. [ergoCubStartupV0.xml](https://github.com/icub-tech-iit/ergocub-software/blob/master/app/ergoCubStartup/ergoCubStartupV0.xml)) but it seems not to match the file installed in the PC which has the following content (and it is not installed by `ergocub-software`):

```
<application>
<name>1-ergoCubStartup</name>

        <dependencies>
        </dependencies>

        <module>
                <name>yarplogger</name>
                <parameters>--start</parameters>
                <node>ergocub000-lap</node>
        </module>

        <module>
                <name>yarprobotinterface</name>
                <parameters>--config ergocub_all.xml</parameters>
                <node>ergocub-torso</node>
        </module>
        
        <module>
                <name>yarpmotorgui</name>
                <parameters>--robot ergocub --parts "(head torso left_arm right_arm left_leg right_leg)"</parameters>
                <node>ergocub000-lap</node>
                <dependencies>
                    <port timeout="60.0" request="is_ready" reply="[ok]">/ergoCubSN000/yarprobotinterface</port>
                </dependencies>
        </module>

</application>
```

Indeed, I was wondering were the startup application is stored (e.g. for other robots it can be found in [robots-configuration](https://github.com/robotology/robots-configuration/tree/master/iCubGenova04/extra/applications)), and if we should update/delete the one stored in this repo.

pattacini, 2023-04-07:
Just for a general remark, XML applications are not stored in robots-configuration. Only certain robots (actually, those AMI has been using the most) contain XML apps.

pattacini, 2023-04-07:
> Just for a general remark, XML applications are not stored in robots-configuration. Only certain robots (actually, those AMI has been using the most) contain XML apps.

To complement this, I have no particular objection to maintaining XML apps in `robots-configuration/ergoCubSN000`.

lrapetti, 2023-04-07:
> > Just for a general remark, XML applications are not stored in robots-configuration. Only certain robots (actually, those AMI has been using the most) contain XML apps.
> 
> To complement this, I have no particular objection to maintaining XML apps in `robots-configuration/ergoCubSN000`.

Since we are mostly using it that way, I am biased towards having the file there. 
Anyway, what I am currently missing is the alternative location since I was not able to find the file anywhere. @pattacini @Nicogene do you know where that file is currently stored?

S-Dafarra, 2023-05-10:
> > Anyway, what I am currently missing is the alternative location since I was not able to find the file anywhere. @pattacini @Nicogene do you know where that file is currently stored?
> 
> Hi @pattacini @Nicogene, if there are no alternatives, I can proceed to commit the file to `robots-configuration/ergoCubSN000`.
> 
> Maybe also @traversaro @S-Dafarra @GiulioRomualdi @mebbaid might be interested in the discussion.

I agree in adding them to an ``extra``` folder in ``robots-configuration/ergoCubSN000``

Nicogene, 2023-05-10:
> Hi @pattacini @Nicogene, if there are no alternatives, I can proceed to commit the file to `robots-configuration/ergoCubSN000`.
>
> Maybe also @traversaro @S-Dafarra @GiulioRomualdi @mebbaid might be [...]

Yes go ahead 👍🏻

lrapetti, 2023-12-04:
The files were added by @GiulioRomualdi in https://github.com/robotology/robots-configuration/pull/592

closed 2023-12-04 by lrapetti