mesh-iit/ergocub-software #91: Develop proper facial expressions handler – Stint 4

Nicogene, 2023-04-12:
### Task description

This is the follow-up of #88 

We would like to:

- [ ] Integrate the new face expression received from rezzonico
- [ ] Use transitions (?)
- [ ] Check the impacto of `ergocubEmotions` on the cpu while for example the realsense is streaming the data

### Definition of done

The above list is completed

cc @pattacini

pattacini, 2023-04-13:
Just received the draft [**`expressions`**](https://github.com/icub-tech-iit/ergocub-software/files/11218437/expressions.zip) from the designers.

pattacini, 2023-04-13:
Moved up to the current sprint as @martinaxgloria will start today.

pattacini, 2023-04-13:
Hi @martinaxgloria 

Please, outline here below how you aim to address the further development required to deal with the transitions, both in terms of the handling of conf files and the necessary C++ objects.

martinaxgloria, 2023-04-13:
Starting from this organization of the conf file which is the one currently used:

```
ergoCubEmotions/
├── config.ini
└── expressions
    ├── images
    │   ├── exp_img_0.png
    │   └── exp_img_1.png
    │   └── ...
    └── videos
        ├── exp_vid_0.mp4
        └── exp_vid_1.mp4
        └── ...
```

it could be an idea to add another folder simply called `transitions` (?) where to store the corresponding videos and it could be organized like:
```
|
└── transitions
        ├── angry-happy.mp4
        ├── neutral-happy.mp4
        ├── shy-angry.mp4
        ├── angry-shy.mp4
        └── angry-neutral.mp4
        └── ...
```
The aim of keeping the name of the emotions instead of `exp_vid_#.mp4` is because it's easier to map the transition between two images/videos from the code side, but I'm open to better ideas.

The `config.ini` file, instead, is currently like this:

```ini
[general]
num_expressions 4

[expression_0]
name happy
type image
file images/exp_img_0.png

[expression_1]
name shy
type image
file images/exp_img_1.png

...
```
From this, we can add a section for each transition with `name`, `type` and `file` sections too, to maintain the same structure.

Let me know if it could work or it's better to move to another direction.

cc @Nicogene @pattacini

pattacini, 2023-04-13:
We need to manage transitions appropriately inside our code, hence we cannot get away simply with `image` and `video` types. Thereby, we're required to have a third type, like `transition`.

Even better, we can have in general `num_expressions` and `num_transitions`, so that we'll have:
```console
[general]
num_expressions 4
num_transitions 10

[expression_0]
//...

[transition_0]
//...
```

An item of the set `transition` doesn't need a `type` attribute (it'll always be a video). Instead, it requires two specific attributes:
- `source`, pointing to the corresponding expression, e.g., `expression_0`.
- `destination`, pointing to the corresponding expression, e.g., `expression_1`.

When reading up the conf file, we need to:
- check if the `source` and `destination` point to available expressions.
- build a further "sort of map" for transitions linking sources to destinations.

Give this structure, the code will follow.

Finally, regarding the file structure, I think we can either stick to the `images` and `videos` sub-directories or you can add up the `transitions` sub-directory, as you like.

martinaxgloria, 2023-04-13:
> Even better, we can have in general `num_expressions` and `num_transitions`

Of course, you're right!

> An item of the set `transition` doesn't need a `type` attribute (it'll always be a video). Instead, it requires two specific attributes:
> 
> * `source`, pointing to the corresponding expression, e.g., `expression_0`.
> * `destination`, pointing to the corresponding expression, e.g., `expression_1`.

You're right about the `type` attribute. Regarding the `source` and `destination`, I was thinking to retrieve them directly from code by storing the command set during the previous iteration and the current setting (this is the reason why I would like to keep the name of the file like `angry-happy`, so it's easier to map them using the ResourceFinder methods like `findFile`). In this way, it's not necessary to build another "sort of map" from source to destination.
Do you think this is not a good approach?

pattacini, 2023-04-14:
We had a meetup w/ Rezzonico yesterday and we decided that:
- We will send them a sample of our OLED.
- They will send us the second batch of expressions+transitions.

pattacini, 2023-04-14:
>We will send them a sample of our OLED.

- https://github.com/icub-tech-iit/tickets/issues/2827
- https://github.com/icub-tech-iit/tickets/issues/2826

martinaxgloria, 2023-04-17:
I pushed on [my fork](https://github.com/martinaxgloria/ergocub-software/tree/feat/stint4) the changes to the module in order to include the transitions from one "static" video to the other one. I've put aside the images for now (they are still present in the configuration files but as comments) and I worked with videos only. 
Tomorrow, I will put here a brief video to show the results.

cc @pattacini @Nicogene

pattacini, 2023-04-17:
Cool!
However, consider that image->transition->image will be most likely our standard pipeline, rather than video->transition->video.

martinaxgloria, 2023-04-17:
Yes, in fact they can be both displayed in the module but since the images we had are very different in quality and resolution with respect to the video received in the past days from Rezzonico, I preferred to have a good _graphic rendering_. Moreover, I wanted to test the video->transition->video which should be the most "tricky" case. 
For the sake of completeness, tomorrow I will do the tests with both cases.

martinaxgloria, 2023-04-18:
Here the result! Sorry for the quality of the video but the videos we have now are overly bright 😅

https://user-images.githubusercontent.com/114698424/232729774-8efef1aa-6d5a-41c4-8956-b4a450bfc33a.mp4

cc @Nicogene @pattacini @maggia80

pattacini, 2023-04-18:
Awesome!

pattacini, 2023-04-18:
The _vademecum_: [vademecum_oled_ergocub.docx](https://github.com/icub-tech-iit/ergocub-software/files/11260490/vademecum_oled_ergocub.docx).

pattacini, 2023-04-18:
@martinaxgloria, I think you can open up the corresponding PR and link it to this issue.

martinaxgloria, 2023-04-18:
PR opened here:

- https://github.com/icub-tech-iit/ergocub-software/pull/94

cc @Nicogene @pattacini

pattacini, 2023-04-18:
Did you already check some borderline conditions such as:
1. `num_transitions` is higher than the actual number of inputs.
2. missing `[transition_#]` group.
3. missing videos and/or files.
4. wrong values for source and/or destination fields.
5. unallowed transition requested via RPC.

martinaxgloria, 2023-04-18:
1. `num_transitions` is only used to fill the corresponding `std::map` in order to associate the "source" and the "destination" emotions.
2. it is related to the first answer.
3. for both images and videos, if the reading results in an empty file, it returns an error message.
4. what do you mean? I didn't get it
5. transitions are not required through RPC, the `setEmotion` service works by passing the expression to be set (for example happy) and inside the code there is a thread that handles the transition between the previous command and the current one.

pattacini, 2023-04-18:
When you process `num_transitions` from the conf file, you ought to parse the conf file to get the information. In this case, if `num_transitions` is for example `5` and you have only `[transition_#]` with `#` in `{0..3}`, you must raise an error.

Likewise, if `num_transitions` is `5` but the conf file contains only `[transition_#]` with `#` in `{0, 1, 3, 4}`, you must raise an error.

>odd values for source and/or destination fields.

A transition has one or both of its fields `source` and/or `destination` pointing to non-existing expressions. In this case, you must raise an error.

>unallowed transition requested via RPC.

You're right! This cannot happen.

pattacini, 2023-04-18:
>Ok, I have to handle this case since I didn't take into account this possibility.

I'd extend this check also to `num_expressions` then.

>For now, if we are in this case, nothing happens. But of course, I can raise an error when it happens.

👍🏻 yep, it'd be a good addition to make the SW more robust.

martinaxgloria, 2023-04-20:
> When you process `num_transitions` from the conf file, you ought to parse the conf file to get the information. In this case, if `num_transitions` is for example `5` and you have only `[transition_#]` with `#` in `{0..3}`, you must raise an error at configuration time.

I was reading again this comment and I have a doubt: if `num_transitions` is higher than the number of the actual inputs (for example, num_transitions = 4 and the inputs are actually 3), it happens that `transitionMap` (which is the std::map that takes care of the transition files) will be filled in the first 3 let's say "positions" and the 4th will be empty.

Instead, if the opposite case occurs, for example num_transitions = 3 and inputs = 4, the transitionMap will be fully filled but one transition remains out since all the map positions are taken.

Based on that, should we handle both cases?

cc @pattacini @Nicogene

pattacini, 2023-04-20:
> if num_transitions is higher than the number of the actual inputs (for example, num_transitions = 4 and the inputs are actually 3), it happens that transitionMap (which is the std::map that takes [...]

This is clearly an error: you have to print out a message and quit the program.

>Instead, if the opposite case occurs, for example num_transitions = 3 and inputs = 4, the transitionMap will be fully filled with the first 3 inputs but one transition remains out since all the map positions are taken.

This is not an error. The only weirdness is that the conf file contains more transitions than those specified in `[general]`. That's ok and it may happen during debugging.

martinaxgloria, 2023-04-20:
> This is clearly an error: you have to print out a message and quit the program.
> 
> >

Clear now, thanks!

pattacini, 2023-04-20:
Just as a general remark, we're required to handle both cases below:
- Expression_1 → Expression_2
- Expression_1 → Transition_12 → Expression_2

martinaxgloria, 2023-04-20:
> When you process `num_transitions` from the conf file, you ought to parse the conf file to get the information. In this case, if `num_transitions` is for example `5` and you have only `[transition_#]` with `#` in `{0..3}`, you must raise an error at configuration time.
> 
> Likewise, if `num_transitions` is `5` but the conf file contains only `[transition_#]` with `#` in `{0, 1, 3, 4}` (i.e., transition `2` is missing), you must raise an error.

About this, instead, I thought to put together these two cases (because it's the same problem but in the first case it happens in the middle, while in the other at the end) and raise an error with the number of the corresponding expression/transition that is missing and then quit the program. What do you think about?

cc @Nicogene @pattacini

pattacini, 2023-04-20:
Agreed 👍🏻 

It's the same case.
You will have to make a for cycle from `0` to `num_transitions-1` and during each iteration `i` you'll quit if you don't find the corresponding `transitions_i`.

martinaxgloria, 2023-04-21:
Today I did a test in order to check the impact of `ergoCubEmotions` while the realsense was streaming data.
I post here the yarpview of both rgb and depth of the realsense while the module is running:

[image]

and the performances of the Jetson Xavier by running `jtop`: asyou can see, the cores are not saturated when the module and the realsense streaming run at the same time.

[image]

cc @pattacini @Nicogene

pattacini, 2023-04-21:
Let's head to the last 2 points:

[image]

pattacini, 2023-04-21:
Regarding the official delivery, I would:
- advertise the functionality in Teams (we have a dedicated channel for ergoCub)
- in the announcement, give availability for in-person explanation/support

martinaxgloria, 2023-04-21:
> Let's head to the last 2 points:
> 
> [image]

For the first point, I will ask @davidelasagna when he's available to do it.

Nicogene, 2023-04-21:
About the official delivery, before doing it I would also add the yarpmanager application on ergocub laptopo in order to make it ready to use(obviously testing that that app works) and I would add some example of change of expressions in the [readme](https://github.com/icub-tech-iit/ergocub-software/blob/master/src/modules/ergoCubEmotions/README.md) using the video we already got. I can add those points to the issue.

Moreover we have to see after removing the service if we can avoid the display the nvidia symbol when the module is not running

martinaxgloria, 2023-04-21:
I update the [README.md](https://github.com/martinaxgloria/ergocub-software/tree/feat/stint4/src/modules/ergoCubEmotions) with a video sample of the module in action.

martinaxgloria, 2023-04-24:
Today, @davidelasagna and I did some tests on ergoCub in order to understand how to deal with the ergoCubEmotion module when the robot starts. After some attempts, in the end, we:

- kept the service that was running on the ergocub-head so far, but we modified it so that only one expression (the "neutral" one) is shown until the module is launched;
- since the service runs at the start, I modified the module by removing the piece of code that did the same thing;
- changed the background with the same image mentioned in the previous point, so that if the module is closed or crashes, an expression continues to be displayed;
- for the same reason as before, we clean up the desktop;
- tried to launch the entire system from the beginning and tried to show the facial expressions successfully.

Now, the last step before the official release is to try to launch the module from the yarpmanager since I didn't have time today, but the application file is already implemented.

I also opened a PR in draft since the last tests have to be done.

- https://github.com/icub-tech-iit/ergocub-software/pull/98

cc @Nicogene @pattacini

pattacini, 2023-04-25:
Nicely done!

>- since the service runs at the start, I modified the module by removing the piece of code that did the same thing;

We shouldn't adapt the module to the service. I'm in favor of leaving it as there wouldn't be a service running underneath. There's absolutely no problem in spotting an image switch when we launch the module.

>- changed the background with the same image mentioned in the previous point, so that if the module is closed or crashes, an expression continues to be displayed;

Don't know if this is desirable. If we close the module or the module itself crashes, there's no problem in seeing the desktop.

Also, in my understanding, the service is still required only for shadowing any pop-up messages while the module is running.

pattacini, 2023-04-26:
>Nope, the service shadows the pop-up windows when the module is not running, so at the start-up.
After that, when the module is running, it takes control of the display in spite of the service and then it's the module that takes care of shadowing the unwanted messages that appear on the screen.

Ok sorry, I didn't catch this point.
Then, I reckon we no longer need the service.

martinaxgloria, 2023-04-28:
After some tests on ergoCub and the integration of the application in yarpmanager, the PR

- https://github.com/icub-tech-iit/ergocub-software/pull/98

is now ready for the review and for the official release.

pattacini, 2023-04-28:
Hi @martinaxgloria 

Once the PR is merged, please fix the cluster XML file in `robots-configuration@master`.
Finally, we can announce that the module is available.

pattacini, 2023-04-28:
>Once the PR is merged, please fix the cluster XML file in robots-configuration@master.

See:
- https://github.com/robotology/robots-configuration/pull/500#pullrequestreview-1405738864

pattacini, 2023-04-28:
Very last point:
- Announcing the delivery on MS Teams.

pattacini, 2023-04-28:
Announcement done! 📣

closed 2023-04-28 by pattacini