mesh-iit/ergocub-software #104: New expressions

pattacini, 2023-05-05:
Hi @martinaxgloria 

Could you please update the expressions with the latest we received from the designers?
Here's the latest delivery: [**`expressions.zip`**](https://github.com/icub-tech-iit/ergocub-software/files/11404779/expressions.zip).

Be careful, the correct tag for one expression is `alert` and not `angry`. Thus, you'd need to update the config.ini file too.

cc @Nicogene @maggia80

martinaxgloria, 2023-05-05:
Hi @pattacini,

yes, I'm already doing all the necessary changes to the module

martinaxgloria, 2023-05-05:
PR here:

- https://github.com/icub-tech-iit/ergocub-software/pull/105

cc @pattacini @Nicogene

pattacini, 2023-05-05:
Reopening just because we'd need a video of the latest expressions running on the robot for comparison purposes, as we aim to get the final @maggia80's approval on that.

martinaxgloria, 2023-05-09:
Today, during the demo session, I took this video in order to check the last version of the facial expressions:

https://github.com/icub-tech-iit/ergocub-software/assets/114698424/3f407843-ac58-4574-a0be-9b8d7a599486

Talking with @maggia80 and @pattacini, we noticed that the videos are not exactly centered on the OLED and maybe we can ask to move up the expressions a little bit more. For the first issue, I will check it more in detail by using one of the expressions with a centered line and then we can send the request to the designers.

martinaxgloria, 2023-05-09:
As expected and as you can see from the photo below, the expressions are not centered at all (the white line is not aligned with the "middle" of the mask)

[image]

cc @pattacini @maggia80 @Nicogene

pattacini, 2023-05-10:
Well done @martinaxgloria 👍🏻 
We need a way to convey the following info to the designers:
1. We need to move the expressions even further up.
2. We need to center the expressions in the middle of the face.Well done @martinaxgloria 👍🏻 
We need a way to convey the following info to the designers:
1. We need to move the expressions even further up.
2. We need to center the expressions in the middle of the face.

To achieve 1 and 2, one possibility is that we stick to one particular expression (the one above is the best, is it `alert`?) and:
1. Draw a horizontal line that meets our requirements: i.e., identify on the robot where the biggest eye should be tangential to that line.
2.  Draw a vertical line that meets the center of the face. With trial and error, you could move the bar further right until the point it is aligned with the groove in the cover.

After a quick internal review, the designer will receive this expression with the two bars and will adjust all the expressions accordingly.

To achieve 1 and 2, one possibility is that we stick to one particular expression (the one above is the best, is it `alert`?) and:
1. Draw a horizontal line that meets our requirements: i.e., identify on the robot where the biggest eye should be tangential to that line.
2.  Draw a vertical line that meets the center of the face. With trial and error, you could move the bar further right until the point it is aligned with the groove in the cover.

After a quick internal review, the designer will receive this expression with the two bars and will adjust all the expressions accordingly.

pattacini, 2023-05-11:
Lovely!
I think we can proceed indeed!
The horizontal bar should be the limit of the right eye's shade.

martinaxgloria, 2023-05-11:
> The horizontal bar should be the limit of the right eye's shade.

Yes, we ask to take this expression as reference and then to adapt the others

pattacini, 2023-05-12:
Here's the reference image: 

[image]

@martinaxgloria sent out the email 📧 
We can close this!

closed 2023-05-12 by pattacini