mesh-iit/ergocub-software #118: Check if changes in the urdf are needed after recent mk2 wrist mech update

Nicogene, 2023-05-19:
### Task description

With the help of @icub-tech-iit/dev we would like to check if the recent changes in the mech of wrist mk2, which is now symmetrical, require updates on urdf side.

### Definition of done

We checked it, and eventually updated it.

cc @pattacini

pattacini, 2023-05-19:
If we need to update the URDF, we can put this task on hold until creo2urdf is ready.
At worst, the required modifications to the virtual serial kinematics are minimal.

cc @Nicogene @salvi-mattia

Nicogene, 2023-06-07:
After a brief chat w/ @salvi-mattia we concluded that since on the wrist the joints are virtual the rotation axis remains the same, the only difference wiil be "aesthetic", the position of the meshes in the urdf could result different but we think that it not worth to update the simprep using the creo9-creo7 procedure for this not-critical change.

The joint limits may change a little bit, but it is just a matter of taking them from the real setup once ready and updating them in the csv, since it is very fast I would not track it in a dedicated issue.

I think we can close it. 

cc @pattacini

pattacini, 2023-06-07:
Agreed 👍🏻

closed 2023-06-07 by Nicogene