mesh-iit/ergocub-software #120: The orientation of the waist_imu_0 frame in the urdf of ergoCubSN000 does not match the frame on the real sensor

isorrentino, 2023-05-24:
Using data from the `waist_imu_0` I noticed that the orientation of the frame `waist_imu_0` in the urdf of `ergoCubSN000` does not match the frame attached to the sensor.

Here, starting from the frame shown in the documentation, I drew the rotated frame on the sensor mounted on `ergoCub`.

Here, starting from the frame shown in the documentation, I drew the frame rotated on the photo of the image mounted on the root_link.

[image]

Here instead, I visualized the frame `waist_imu_0` using the `ergoCub` urdf.

[image]

Here is the `matlab` script I used 

<details><summary>Click here to see the code.</summary>

```MATLAB
%% Settings

robotName='ergoCubSN000'; %% Name of the robot

testFrames = {'waist_imu_0'}; %% The frames to display

modelPath = '';

fileName='model.urdf'; %% Name of the urdf file

jointOrder={     %% The list and the order of the joints
    'r_hip_pitch';
    'r_hip_roll';
    'r_hip_yaw';
    'r_knee';
    'r_ankle_pitch';
    'r_ankle_roll';
    'l_hip_pitch';
    'l_hip_roll';
    'l_hip_yaw';
    'l_knee';
    'l_ankle_pitch';
    'l_ankle_roll';
    'torso_pitch';
    'torso_roll';
    'torso_yaw';
    'r_shoulder_pitch';
    'r_shoulder_roll';
    'r_shoulder_yaw';
    'r_elbow';
    'r_wrist_prosup';
    'r_wrist_pitch';
    'r_wrist_yaw';
    'l_shoulder_pitch';
    'l_shoulder_roll';
    'l_shoulder_yaw';
    'l_elbow';
    'l_wrist_prosup';
    'l_wrist_pitch';
    'l_wrist_yaw';
    'neck_pitch';
    'neck_roll';
    'neck_yaw'
    };

groundFrame = 'l_sole'; %% The frame where to place the ground

rootLink = 'root_link'; %% The base link

joints_positions=zeros(length(jointOrder),1); %% The joints'position

printModel = true; %% Define if the model has to be printed on the command window

%% No need to edit from here

% Main variable of iDyntreeWrappers used for many things including updating
% robot position and getting world to frame transforms
KinDynModel = iDynTreeWrappers.loadReducedModel(jointOrder, rootLink, modelPath,fileName,false);

if printModel
    KinDynModel.kinDynComp.model().toString() %%Print model
end

% Set initial position of the robot
iDynTreeWrappers.setRobotState(KinDynModel,eye(4),joints_positions,zeros(6,1),zeros(size(joints_positions)),[0,0,-9.81]);

groundFrameTransform = iDynTreeWrappers.getWorldTransform(KinDynModel, groundFrame);

world_H_base = groundFrameTransform^-1;

iDynTreeWrappers.setRobotState(KinDynModel,world_H_base,joints_positions,zeros(6,1),zeros(size(joints_positions)),[0,0,-9.81]);

% Prepare figure, handles and variables required for the update, some extra
% options are commented.
[visualizer,objects]=iDynTreeWrappers.prepareVisualization(KinDynModel,meshFilePrefix,...
    'color',[1,1,1],'transparency',1, 'name', ['Plot frame ', robotName], 'reuseFigure', 'name');
xlim([-0.5, 0.5])
ylim([-0.5, 0.5])
zlim([-0.05, 1.4])

for t = 1 : length(testFrames)
    frameTransform = iDynTreeWrappers.getWorldTransform(KinDynModel, testFrames{t});
    testFramesObjects{t} = iDynTreeWrappers.plotFrame(frameTransform, 0.2, 5);
end
```
</details>

cc @GiulioRomualdi

martinaxgloria, 2023-05-29:
I open a PR here 

-  https://github.com/icub-tech-iit/ergocub-software/pull/121

cc @Nicogene @isorrentino

Nicogene, 2023-05-30:
@isorrentino if you confirm that now is fixed we can close this issue, thanks!

isorrentino, 2023-05-30:
@Nicogene @martinaxgloria now it is correct, see here

[image]

Thank you all.

closed 2023-05-30 by Nicogene