mesh-iit/ergocub-software #129: Wrong data from gyroscopes and accelerometers

isorrentino, 2023-06-05:
By checking data from the robot I discovered that the gyroscope and accelerometer measurements do not match the values computed from forward kinematics. In particular, all the axes have the sign inverted and the `x` and `y` axes are switched.

Here is the script I used for these plots and a [dataset](https://istitutoitalianotecnologia-my.sharepoint.com/:u:/g/personal/ines_sorrentino_iit_it/EYyQZKmKIz9FuKSWE3q2y-oBVuIJ5SPMxP2UFnabfW_e-g?e=xYbWsB).

[check_imu_frame.zip](https://github.com/robotology/icub-tech-support/files/11654292/check_imu_frame.zip)

The figures compare the measurements from the sensors (in blue) and the forward kinematics (in red). 

While the plot on the left side shows the measurements from the sensors as they are, the one on the right shows the axes switched as:

```
x = -y
y = -x
z = -z
```

### Gyroscope right foot front
[image]

### Accelerometer right foot front
[image]

I checked also the right foot rear, but not the other sensors.

cc @GiulioRomualdi @traversaro @Nicogene @pattacini

traversaro, 2023-06-05:
@isorrentino I am not wrong this data was collected on the ergoCubSN000 before leaving from London and updating to the last firmware, so we are not sure which version of the firmware was running on the STRAIN2, right?

traversaro, 2023-06-05:
Something that I noticed (and I do not know if it is related) is that https://github.com/robotology/icub-firmware/pull/313 added a new board called `strain2c` . I am not sure where that board is used (a search https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20strain2c&type=code does not give any result). However, apparently that board uses a macro called `STM32HAL_BOARD_STRAIN2C`, but the logic to handle the different frame used by bno055 on strain2 only checks for `STM32HAL_BOARD_STRAIN2`, see https://github.com/robotology/icub-firmware/blob/5cf9dd1965140a5088d8fd41a41b1132a8fffe1e/emBODY/eBcode/arch-arm/embot/app/embot_app_application_theIMU.cpp#L526 .

traversaro, 2023-06-05:
> But how did the robot manage to walk in London?
> Doesn't walking make use of IMU data? Probably not...

The walking mainly uses the xsens imu in the pelvis, I guess the issue is w.r.t. to an IMU mounted on an FT sensor.

@isorrentino can you share the name of the IMU that is missbehaving? Thanks!

traversaro, 2023-06-05:
> @isorrentino can you share the name of the IMU that is missbehaving? Thanks!

Found, it is the `r_foot_front_ft`.

isorrentino, 2023-06-05:
> @isorrentino can you share the name of the IMU that is missbehaving? Thanks! 

I checked the ones mounted on the `right_foot_front` and the `right_foot_rear` ft sensors. The names should be `r_foot_front_ft_acc` and `r_foot_rear_ft_acc`. or without `acc`.

GiulioRomualdi, 2023-06-05:
On the 18th of May I flashed the firmware of the 2foc as written in https://github.com/robotology/robots-configuration/pull/522

I used this tag: https://github.com/robotology/icub-firmware-build/releases/tag/v1.34.1

pattacini, 2023-06-05:
> Something that I noticed (and I do not know if it is related) is that [robotology/icub-firmware#313](https://github.com/robotology/icub-firmware/pull/313) added a new board called `strain2c` . I am not sure where that board is used (a search https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20strain2c&type=code does not give any result). However, apparently that board uses a macro called `STM32HAL_BOARD_STRAIN2C`, but the logic to handle the different frame used by bno055 on strain2 only checks for `STM32HAL_BOARD_STRAIN2`, see https://github.com/robotology/icub-firmware/blob/5cf9dd1965140a5088d8fd41a41b1132a8fffe1e/emBODY/eBcode/arch-arm/embot/app/embot_app_application_theIMU.cpp#L526 .

Probably this is the point 👍🏻 

cc @Nicogene @marcoaccame @davidetome

marcoaccame, 2023-06-06:
> > Something that I noticed (and I do not know if it is related) is that [robotology/icub-firmware#313](https://github.com/robotology/icub-firmware/pull/313) added a new board called `strain2c` . I am not sure where that board is used (a search https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20strain2c&type=code does not give any result). However, apparently that board uses a macro called `STM32HAL_BOARD_STRAIN2C`, but the logic to handle the different frame used by bno055 on strain2 only checks for `STM32HAL_BOARD_STRAIN2`, see https://github.com/robotology/icub-firmware/blob/5cf9dd1965140a5088d8fd41a41b1132a8fffe1e/emBODY/eBcode/arch-arm/embot/app/embot_app_application_theIMU.cpp#L526 .
> 
> Probably this is the point; good catch👍🏻
> 
> cc @Nicogene @marcoaccame @davidetome

i don't think that the `strain2c` is used yet. But yes: the rotation must be applied also if the macro `STM32HAL_BOARD_STRAIN2C` is defined. However, it may be even better to impose the rotation somewhere else deep inside the HW driver.

pattacini, 2023-06-06:
>However, it may be even better to impose the rotation somewhere else deep inside the HW driver.

The conversion is implemented to realign the IMU frame to the FT frame so the driver shouldn't be involved in that operation, I think, as it should be rather responsible only for dealing with the IMU itself.

[^1]: See https://icub-tech-iit.github.io/documentation/ft_sensors/ft_onboard_imu/#measurement-frame-specifications

Nicogene, 2023-06-06:
The robot will be back soon and we might double-check the strain2 fw version.

Before that, we might check if the forces/accelerations/gyro are expressed in the same reference system on w/ an FT on the bench.

For doing it, using https://github.com/icub-tech-iit/ft-sensors-simulink could ease the test

cc @pattacini @davidetome @marcoaccame

pattacini, 2023-06-06:
Also, we could already patch the FW by dealing with the macro `STM32HAL_BOARD_STRAIN2` as well.
We should be fairly quick in view of the distro, hence other refactorings could come later.

davidetome, 2023-06-07:
added check for `strain2c` macro for IMU : 

- https://github.com/davidetome/icub-firmware/commit/97b5b64c935a25d79a137ba21f84f94f9b2513b5

Tomorrow I'll test other stuff

cc @Nicogene @marcoaccame

davidetome, 2023-06-08:
Today I checked the axis map using the [Simulink model](https://github.com/icub-tech-iit/ft-sensors-simulink ), here are the results : 

## F/T sensor w/ the latest FW in devel
The mapping is : 
````
x = y
y = x
z = -z
````
## F/T sensor w/ the latest FW in devel without the remapping
Removing the macro : 

```cpp
#if defined(STM32HAL_BOARD_STRAIN2)
    // In strain2 we have the P6 configuration reported in bno055 datasheet
    embot::hw::bno055::write(pImpl->config.sensor, embot::hw::bno055::Register::AXIS_MAP_CONFIG, 0x21, 5*embot::core::time1millisec);
    embot::hw::bno055::write(pImpl->config.sensor, embot::hw::bno055::Register::AXIS_MAP_SIGN, 0x07, 5*embot::core::time1millisec);
#endif
````

Yhe mapping is again : 

````
x = y
y = x
z = -z
````

## F/T sensor w/ the FW included in the v1.26.0 release
Using the FW released on 31/8/22 when the remapping was implemented

The mapping is : 

````
x = -x
y = -y
z =  z
````

ℹ️ The difference between the 2 FW is the STM32HAL lib used and some mods in the driver. We've to deeply inspect these differences

cc @Nicogene @marcoaccame @pattacini

Nicogene, 2023-06-12:
After a brief alignment w/ @marcoaccame and @davidetome we decided to check if we request runtime of changing the reference frame the measure changes or it is ignored in order to narrow-down the problem to eventually the startup configuration

[image]

davidetome, 2023-06-13:
Thoi morning I tested sending the CAN message to set orientation P5 : 

[image]

And the orientation of the axis changes (in this case Z axis)

[image]
[image]

cc @Nicogene @marcoaccame

Nicogene, 2023-06-13:
We should have found the bug, we already committed the fix, we should test it first:

- https://github.com/robotology/icub-firmware/pull/390
- https://github.com/robotology/icub-firmware-build/pull/89

cc @marcoaccame @pattacini @davidetome @isorrentino @traversaro

traversaro, 2023-06-13:
Great, thanks a lot!

traversaro, 2023-06-13:
Do you have an idea of what caused the regression and which versions of the firmware are affected?

Nicogene, 2023-06-13:
> Do you have an idea of what caused the regression and which versions of the firmware are affected?

This PR:
- https://github.com/robotology/icub-firmware/pull/347

Added the possibility of configuring the IMU reference frame via CAN, and it came after

- https://github.com/robotology/icub-firmware/pull/286

So the versions:
- `2.0.16`
- `2.1.0`

Are affected by this bug as the ones < `2.0.15`.

It was due to the fact that if not specified in the imu config message, the placement `P1` was set by default, instead of keeping the same set at bootstrap (that for `strain2` was `P6`)

traversaro, 2023-06-13:
> It was due to the fact that if not specified in the imu config message, the placement `P1` was set by default, instead of keeping the same set at bootstrap (that for `strain2` was `P6`)

Thanks, this is now clear!

davidetome, 2023-06-15:
Orientation is now ok 👍🏻 

cc @Nicogene @marcoaccame

Nicogene, 2023-06-16:
Fixed by:

- https://github.com/robotology/icub-firmware-build/pull/89
- https://github.com/robotology/icub-firmware/pull/390

DanielePucci, 2023-06-18:
Hi all, thanks for dealing with this. I was wondering: what is the plan to update the FT sensors on the ergoCubSN000?

Nicogene, 2023-06-20:
Hi @DanielePucci, I am in contact w/ @AntonioConsilvio  for updating the FT of the robot since it is in Rome, worst case we will update it when it will be back

DanielePucci, 2023-06-21:
Thanks for the feedback @Nicogene, looks a nice plan

DanielePucci, 2023-10-03:
Cc @g-cervettini