mesh-iit/ergocub-software #131: Wrong lidar mounting on ergocubSN000

SimoneMic, 2023-06-14:
Hello,

Today we've tested the lidar on the robot and we have seen a mismatch between the mounting of the lidar with the URDF of the robot.
In few words: the lidar should be rotated by 90 degrees clockwise.

Here's a screenshot of the lidar on Rviz, the readings circled in blue are the ones crane and the support of the robot that should be behind. Here they are on the right of the robot (the red axis in Rviz is the x-axis that points forward)

[image]

[image]

Here we tried to put a box behind the robot and is still displayed to the side
[image]

Do we change the URDF of the robot or fix the mounting of the lidar?
The easiest way is to adapt the URDF, but then it should be consistent through all the other versions of ergocub and also the model on gazebo

pattacini, 2023-06-14:
Thanks for reporting 👍🏻 

>Do we change the URDF of the robot or fix the mounting of the lidar?
The easiest way is to adapt the URDF, but then it should be consistent through all the other versions of ergocub and also the model on gazebo

We need to check which is the intended mounting and then adapt accordingly.

pattacini, 2023-06-15:
From a first inspection, it came out that the LIDAR is mounted correctly as per the CAD.
The problem should be then a missing/wrong CSYS that impacted the URDF.

Stay tuned.

Nicogene, 2023-06-15:
This issue is confirmed by the fact that right now this is the frame exported

[image]

But the datasheet says that the x is ahead

[image]

randaz81, 2023-06-15:
No, the lidar is right-handed.
The x is pointing forward, y on the left, z is pointing upward. Scans dara are ordered CCW.
As mentioned by @SimoneMic just a 90 deg rotation is fine to fix the issue.

randaz81, 2023-06-15:
What is shown in the datasheet picture mentioned above should be considered wrong and is not consistent with our yarp/ros data representation.

pattacini, 2023-06-16:
Hi @randaz81 

It's true that a left-handed triplet is kind of unusual/weird but could you elaborate more on the following?

> What is shown in the datasheet picture mentioned above should be considered wrong and is not consistent with our yarp/ros data representation.

To which extent should the datasheet be considered misleading?

randaz81, 2023-06-16:
Because it states: "The rotation angle increases as rotating clockwise".

However, we are representing the scan data ordered _counterclockwise_.
In fact, giving an X pointing to the front of the robot, and Y pointing to the left, then Z is pointing upwards (right-handed).
And Z pointing upwards defines a _counterclockwise_ rotation on a right-handed reference system (which is ok, while a clockwise rotation would be left-handed).

Extras about ROS standards:
https://www.ros.org/reps/rep-0103.html
https://www.ros.org/reps/rep-0105.html

pattacini, 2023-06-16:
>Because it states: "The rotation angle increases as rotating clockwise".

This is expected for a left-handed triplet, I think.

Theoretically, we ought to insert in the CAD the information relative to the physical device to then deal with it at a higher level.

Nonetheless, there are a bunch of things to consider as well:
- The flexibility of the CAD to deal with left-handed CSYS.
- The possibility to have left-handed within a URDF (probably there ain't any).

cc @fiorisi @Nicogene @maggia80

randaz81, 2023-06-16:
It's possible that the datasheet represents how the device internally works, i.e. the direction of rotation of the motor (for the S2 device it is hidden inside the case so it cannot be seen from outside). However, the SDK provides an array of tuples [angle, distance] in random order (for example, the angle could be: 3, 4, 7, 5, 2..] and is up to the user to process it. However, I don't think that is necessary to know these internal details exactly: given the fact the device has axial symmetry, the x axis, which defines the 0 angle, is defined merely by the software, and it can be placed (in our device driver code) as the user prefers. In our case, we are publishing data following the ROS convention, and what the datasheet says can be completely ignored. Please note that this consideration of mine is specific for a 2D lidar device, which has z-axis symmetry, but it is not general. Indeed, for an RGB camera or an IMU, the reference frame orientation should be chosen carefully according to the hardware.

pattacini, 2023-06-20:
Hi @Mick3Lozzo 

Let's proceed with a right-handed CSYS with the x-axis and the y-axis oriented as required by the discussion above.

Mick3Lozzo, 2023-06-20:
[image]

[image]

Added the coordinate system `CSYS_RPLIDAR_2` in the RPLIDAR commercial part, as a reference for `SCSYS_HEAD_S2_LASER` at an assembly level. Now the orientation is compliant with the standard.

@randaz81 @pattacini @mfussi66 @Nicogene

Nicogene, 2023-06-20:
If now is ok for @randaz81 and @SimoneMic we can close it.

PR #134 (2023-06-15): Fix lidar reference frame
It fixes #131 

We fixed the frame of the lidar now the X point ahead as the other frames in the robot

[image]

We applied the patch on generation side and we didn't fixed the CAD for the issue of creo 9

cc @SimoneMic @pattacini
closed 2023-06-21 by Nicogene