mesh-iit/ergocub-software #137: Cannot build the librealsense on ergoCubSN000

xEnVrE, 2023-06-16:
As requested by @Nicogene, I am posting here the errore that we currently get when trying to build the current `librealsense` installation within the `ergocub-head` machine:

Steps to reproduce:

```console
ergocub@ergocub-head:~$ cd $ROBOT_CODE/librealsense/build
ergocub@ergocub-head:~$ make
```

The output that we get:

```console
[  3%] Built target fw
[ 12%] Built target realsense-file
[ 13%] Building CUDA object CMakeFiles/realsense2.dir/src/cuda/cuda-pointcloud.cu.o
/home/ergocub/mambaforge/envs/ros_humble/aarch64-conda-linux-gnu/include/c++/11.3.0/bits/std_function.h:435:145: error: parameter packs not expanded with '...':
  435 |         function(_Functor&& __f)
      |                                                                                                                                                 ^ 
/home/ergocub/mambaforge/envs/ros_humble/aarch64-conda-linux-gnu/include/c++/11.3.0/bits/std_function.h:435:145: note:         '_ArgTypes'
/home/ergocub/mambaforge/envs/ros_humble/aarch64-conda-linux-gnu/include/c++/11.3.0/bits/std_function.h:530:146: error: parameter packs not expanded with '...':
  530 |         operator=(_Functor&& __f)
      |                                                                                                                                                  ^ 
/home/ergocub/mambaforge/envs/ros_humble/aarch64-conda-linux-gnu/include/c++/11.3.0/bits/std_function.h:530:146: note:         '_ArgTypes'
make[2]: *** [CMakeFiles/realsense2.dir/build.make:1549: CMakeFiles/realsense2.dir/src/cuda/cuda-pointcloud.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:798: CMakeFiles/realsense2.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

I see here that the `c++` compiler is picked from the `mambaforge` environment. Not sure if this is normal though.

cc @Nicogene

Nicogene, 2023-06-16:
Thanks @xEnVrE for letting us know, we have to investigate it w/ @martinaxgloria 

cc @pattacini

martinaxgloria, 2023-07-10:
Hi @xEnVrE,

tomorrow I succeeded in reproducing your error while compiling librealsense. Since in `ergocub-head` the conda environment named `ros_humble` is active by default, you were not able build it because of some missing dependencies. To solve the problem, I first run `conda deactivate` to act out of the environment. Then, I deleted the previous build folder and created a new one, I generated again the CMake files, and then the building process successes. Finally, I reactivate conda env with `conda activate ros_humble`.

cc @Nicogene

xEnVrE, 2023-07-10:
Thank you, this information will be very useful for future reference.

Can I ask if you also recompiled the [yarp-device-realsense2](https://github.com/robotology/yarp-device-realsense2) device after that and verified that the camera runs succesfully?

Thank you

martinaxgloria, 2023-07-10:
> Can I ask if you also recompiled the [yarp-device-realsense2](https://github.com/robotology/yarp-device-realsense2) device after that and verified that the camera runs succesfully?

Unfortunately, my reservation time was over, I will do it this afternoon. I will let you know!

martinaxgloria, 2023-07-10:
I rebuilt `yarp-device-realsense2` and I succeeded in launching the realsense views as you can see in the picture below:

[image]

cc @xEnVrE @Nicogene

closed 2023-07-10 by Nicogene