mesh-iit/ergocub-software #142: Unreliable ROS2 topic detection from `ergocub000-lap`

SimoneMic, 2023-06-27:
Hello,

In brief: while running ros2 nodes on different `ergoCub` PCs, we are not able to reliably see them from the laptop. Sometimes ros2 is not able to detect all the topics or, if some topics are detected, is not able to read the content.

Detailed explanation: We have different nodes running on different platforms:
* torso: we have the ros2 remappers of the joints and force sensors, also the robot state publisher - `/joint_states` `/tf` `/force_sensors`
* head: the scan and realsense - `/scan` `/camera...`
* laptop: where the navigation code runs

From the torso I am able to reliably see and echo all the topics on the head and vice versa. On the laptop I can't, nor the torso nor the head topics.
If I start a dummy publisher on the laptop it cannot be seen on the head/torso and the same way around, confirming the phenomenom.

I have also noticed that all the PCs are using `rmw_fastrtps_cpp` instead of `Cyclone`.
Also the ROS_DOMAIN_ID is the same on all the three machines.

Another funny behavior is that if I try to diagnose repeatedly on the same terminal, on the same machines, I have different results on the available publishers.

FYI @randaz81 @vigisushrutha23

--LOGS BELOW--

If I try to diagnose the state of the ros2 network here's what I get (I have omitted the QoS checks since are all OK):
LAPTOP
```
ergocub@ergocub000-lap:/usr/local/src/robot/ros2_ws/install$ ros2 doctor --report
WARNING: Package name "iCub" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/__init__.py: 154: UserWarning: Fail to call PackageReport class functions.

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,UP,RUNNING>
mtu          : 65536
inet         : 10.0.2.104
inet4        : ['10.0.2.104']
ether        : 74:04:f1:fb:51:cb
inet6        : ['fe80::afad:1af5:e942:d8fc%wlo1']
netmask      : 255.255.255.0
device       : wlo1
flags        : 4163<RUNNING,UP,BROADCAST,MULTICAST>
mtu          : 1500
broadcast    : 10.0.2.255
inet         : 172.17.0.1
inet4        : ['172.17.0.1']
ether        : 02:42:bc:2c:8e:2f
netmask      : 255.255.0.0
device       : docker0
flags        : 4099<UP,BROADCAST,MULTICAST>
mtu          : 1500
broadcast    : 172.17.255.255
ether        : ac:91:a1:06:bb:8e
device       : enp45s0
flags        : 4099<UP,BROADCAST,MULTICAST>
mtu          : 1500

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.19.0-45-generic-x86_64-with-glibc2.35
release          : 5.19.0-45-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
All OK

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

[... 227 lines cut ...]
topic               : /left_foot_heel_tiptoe_ft
publisher count     : 1
subscriber count    : 0
topic               : /right_arm_ft
publisher count     : 1
subscriber count    : 0
topic               : /right_foot_heel_tiptoe_ft
publisher count     : 1
subscriber count    : 0
topic               : /robot_description
publisher count     : 1
subscriber count    : 0
topic               : /scan
publisher count     : 1
subscriber count    : 0
topic               : /string_topic
publisher count     : 1
subscriber count    : 0
topic               : /tf
publisher count     : 1
subscriber count    : 0
topic               : /tf_static
publisher count     : 1
subscriber count    : 0
```

randaz81, 2023-06-29:
I never noticed this bizarre behavior on R1 setup. What do you think @elandini84?

@SimoneMic maybe we could also test the new "iron" ROS2 distribution instead of humble. We already tested it on R1.

randaz81, 2023-06-29:
> I have also noticed that all the PCs are using rmw_fastrtps_cpp instead of Cyclone.

No please use Cyclone! If I remember correctly fast dds is known to be bugged.... that could be the issue!

elandini84, 2023-06-29:
> I never noticed this bizarre behavior on R1 setup. What do you think @elandini84?
> 
> @SimoneMic maybe we could also test the new "iron" ROS2 distribution instead of humble. We already tested it on R1.

I've never seen a behaviour like this on R1 either. I also agree on replacing fast dds.

SimoneMic, 2023-06-29:
I will install Cyclone as soon as I have the robot available, however I am not sure on how to install it on the head, since ros2 was installed on mambaforge or another environment handler.

Nicogene, 2023-06-30:
Hi @SimoneMic, on `ergocub-head` we are using mamba because since it is `Ubuntu 20.04` ros2 is stuck to `foxy` if you install it via apt.

I don't know how to change dds under the hood, if it is just needed to install another package, you need to check if it is available via conda (cc @traversaro).

In any case, since the robot is a shared resource I would do this test first on a Xavier on bench (if you need it we have it one on first floor, you can ask me), because also yarp and the realsense device relies on conda, so there is the possibility of breaking the AMI workflow.

And then this change has to be made also on `ergocub-laptop` and `ergocub-torso`, in order to have the whole system aligned. (On those ros2 is installed w/ apt, since they have Ubuntu 22.04).

cc @pattacini @martinaxgloria

SimoneMic, 2023-06-30:
On my docker I install Cyclone DDS using these commands:

```
RUN sudo apt-get update -y
RUN sudo apt-get install -y cyclonedds-dev cyclonedds-tools ros-humble-cyclonedds-dbgsym ros-humble-cyclonedds ros-humble-rmw-cyclonedds-cpp ros-humble-rmw-cyclonedds-cpp-dbgsym libcycloneddsidl0
#Run in every terminal the alias fast_dds_setup
ENV RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```
So I think that an alternative could be to install it via source code as said from the [installation instructions](https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html)
I think that we should aim to have the same version on all the systems if possible. Do you have any suggestion @randaz81 ?

traversaro, 2023-06-30:
> > I have also noticed that all the PCs are using rmw_fastrtps_cpp instead of Cyclone.
> 
> No please use Cyclone! (`rmw_cyclonedds_cpp`) If I remember correctly fast dds is known to be bugged (at least one year ago).... that could be the issue!

Indeed, in general on ROS2 no compatibility is guaranteed when using different dds vendors, see https://docs.ros.org/en/humble/Concepts/Intermediate/About-Different-Middleware-Vendors.html#multiple-rmw-implementations :

> For many cases you will find that nodes using different RMW implementations are able to communicate, however this is not true under all circumstances. Here is a list of inter-vendor communication configurations that are not supported:

pattacini, 2023-06-30:
Hi @SimoneMic 

>So I think that an alternative could be to install it via source code as said from the [installation instructions](https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html)

This could be a viable option, although I don't know how it integrates with the mamba infrastructure.

>I think that we should aim to have the same version on all the systems if possible. 

Baer in mind that the GPU equipping the head poses some limitations in terms of supported OS and thus SW versions.

traversaro, 2023-06-30:
> This could be a viable option, although I don't know how it integrates with the mamba infrastructure.

If ROS2 is installed via conda/mamba, then the `ros-humble-cyclonedds` package is available  (see https://robostack.github.io/humble.html), except for Windows.

SimoneMic, 2023-07-04:
I have installed Cyclone DDS on all the three machines and I've configured it forbidding multicast outside of the robot environment.
A file example (for the laptop) is the following:
```
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
   <Domain Id="any">
      <General>
         <AllowMulticast>false</AllowMulticast>
         <Interfaces>
            <NetworkInterface address="10.0.2.4" priority="default"/>
         </Interfaces>
      </General>
      <Discovery>
         <MaxAutoParticipantIndex>30</MaxAutoParticipantIndex>
         <ParticipantIndex>auto</ParticipantIndex>
         <Peers>
            <Peer address="10.0.2.4"/>
            <Peer address="10.0.2.2"/>
            <Peer address="10.0.2.3"/>
         </Peers>
      </Discovery>
   </Domain>
</CycloneDDS>
```
Now I am able to see ros2 across the PCs. However, I have found that the date of the three PCs was not synchronized, mostly the torso had a huge offset of:
```
ergocub@ergocub-torso:~$ sudo ntpdate -b 10.0.2.1
 4 Jul 15:11:50 ntpdate[3875]: step time server 10.0.2.1 offset -7247.956493 sec
```

To avoid running `ntpdate `manually each time on all the machines, I have followed [this guide](https://www.linuxbabe.com/linux-server/how-to-enable-etcrc-local-with-systemd). On the laptop I have enabled the `rc-local.service` to run `ntpdate -b 10.0.2.1` at startup.
Unfortunately I don't have permissions to do that on `ergocub-torso` nor `ergocub-head` @davidelasagna

SimoneMic, 2023-07-07:
`ergocub-torso` is still lagging behind from the other two PCs by around 7000 seconds, when updating the first time with `ntpdate -b 10.0.2.1`. It seems that it's not retaining the date after the power off

xEnVrE, 2023-08-18:
Just wanted to add that these days me and @fbrand-new were experiencing similar issues on `iCubGenova11` when using the default ROS2 middleware, i.e. `rmw_fastrtps_cpp`. We could not receive topics published on `icub-head` on other laptops connected to the network, or at least not always. 

At some point we have realized that the problem is that **when we turn on the robot motors,** a new network interface activates on the head and gets assigned an ip address - the one used to communicate to the motor control boards. After that, **any new ROS2-based application launched on the head was streaming data to that interface, the wrong one.**

To solve that we added an xml configuration as follows:
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>CustomUDPTransport</transport_id>
            <type>UDPv4</type>
            <interfaceWhiteList>
                <address><NIC_ip></address>
            </interfaceWhiteList>
        </transport_descriptor>
    </transport_descriptors>
    <participant profile_name="CustomUDPTransportParticipant" is_default_profile="true">
        <rtps>
            <userTransports>
                <transport_id>CustomUDPTransport</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
</profiles>
```
where `<NIC_ip` has to be filled and we let ROS 2 aware of this change with the environment variable `FASTRTPS_DEFAULT_PROFILES_FILE=<path_to_xml_file>`.

Just to say that maybe even the default RMW is working properly, it just a matter of using the correct network interface.

Maybe this was the reason behind the strange behavior described in https://github.com/icub-tech-iit/ergocub-software/issues/142#issue-1777349455?

traversaro, 2023-08-18:
Cool debug @xEnVrE @fbrand-new ! If in the future we experience again this strange behaviour, it would be cool to dump the networ traffic on the head, so that we can see that ROS2 packets are indeed sent (I guess in broadcast) to the wrong network interface. A simple way to dump the data on all the interfaces is `sudo tcpdump -i any`. The resulting .pcap file can be opened easily by WireShark.

traversaro, 2023-08-19:
Perhaps https://github.com/ros2/rmw_fastrtps/issues/593 and https://github.com/ros2/rmw_fastrtps/issues/609 could be related issues.

SimoneMic, 2023-11-16:
Forgot to close this issue, sorry.

It has been solved by using [this file](https://github.com/SimoneMic/ergocub_navigation/blob/main/config/cyclonedds.xml) (on each PC with its proper IP setup) and automatically synchronizing all the PCs at startup (and/or manually by `sudo ntpdate -b 10.0.2.1`)

We don't have great delays in this situation, also we avoid publishing a great load of data on wifi via ROS2

closed 2023-11-16 by SimoneMic