mesh-iit/ergocub-software #148: Evaluation on Using ROS2 Iron on `ergoCubSN000`

SimoneMic, 2023-07-13:
Hello,

As per title, since ROS2 Iron (the latest release) has been released I think that we should evaluate the possibility to replace ROS2 Humble.
[Here](https://docs.ros.org/en/iron/Releases/Release-Iron-Irwini.html) there are the release notes with the changes since Humble.
There are some pros, like the improved and managed discoverability of topics (similar to the issue that we had #142 ).
The con that I can think of is that it's not a long term supported version, unlike Humble. Comparison [here](https://docs.ros.org/en/iron/Releases.html).

Also since the HSP Navigation team is moving to Iron, it could be helpful to have all the projects in CRIS aligned on this. cc @randaz81
What would be the impact on switching to Iron on the actual robot?

SimoneMic, 2023-07-14:
To avoid the head issue we could exclude it completely from having ros2.
I don't know if in the future it could be useful to still have it, but I am actually using yarp to stream the lidar and camera data from the head to the laptop, and here convert it to ros2.
The only other sensor that could be helpful to have is the head IMU, but if we can keep the same process above it should be fine.

So we could remove ros2 from the head and keep it only on the torso and the laptop.
Or could this create some future issues? 
cc @randaz81

traversaro, 2023-07-14:
@GiulioRomualdi in the past used ergocub-head/realsense with ROS2, but I am not sure what it is the status now.

randaz81, 2023-07-14:
I discussed this issue with f2f with @Nicogene and @pattacini. We concluded that it would be optimal to schedule a f2f meeting with also @traversaro and @GiulioRomualdi to check the impact on all groups about migrating to ROS2 iron. 
- In general switching the whole setup (robot/laptop) to iron is a mandatory step, given the fact that humble is affected by some bugs. On R1 we are already using iron.
- The solution proposed by @SimoneMic about using yarp nwc/nws to transfer data without involving ROS2 in the head is very good and I would definitely go for it. (This is not a workaround, it is a viable and efficient solution!)

traversaro, 2023-08-03:
From what I understand from @pattacini , the plan is now to update to ROS2 Iron, switching to do the installation with apt and not with conda. However, something that I missed is what is the plan with `ergocub-head` that has 20.04, the possible options I see are:
* Do not install ros2 in ergocub-head
* Install it via Docker on 22.04 

What was the decision w.r.t. to this point?

pattacini, 2023-08-04:
For what regards the head, I think that @randaz81 and @SimoneMic will rely on NWS, which also allows for some data compression.

traversaro, 2023-08-04:
Ok, so I guess ROS2 will not be installed in ergocub-head.

SimoneMic, 2023-08-24:
Actually I am testing some packages like `robot_localization` that have a stable version based on humble. While there's a rolling version supporting Iron, I want to test before the stable version first. So I'll wait a bit to start switching the versions.