mesh-iit/ergocub-software #149: wholebodydynamics does not start with ergoCubGazeboV1

isorrentino, 2023-07-18:
Here is the error

```bash
[INFO] |yarp.dev.PolyDriver|multipleanalogsensorsremapper| Created device <multipleanalogsensorsremapper>. See C++ class MultipleAnalogSensorsRemapper for documentation.
[DEBUG] wholeBodyDynamics Statistics: Multiple analog sensors remapper opened in  0.009 s.
[DEBUG] wholeBodyDynamics Statistics: Configuration finished. Waiting attachAll to be called.
[INFO] |yarp.dev.PolyDriver|wholebodydynamics| Created device <wholebodydynamics>. See C++ class yarp::dev::WholeBodyDynamicsDevice for documentation.
[INFO] Entering action level 15 of phase startup
[INFO] Executing attach action, level 15 on device wholebodydynamics with parameters [("networks" = "(left_leg right_leg torso right_arm left_arm imu waist_imu left_arm_ft_sensor right_arm_ft_sensor left_leg_ft_sensor right_leg_ft_sensor)"), ("left_leg" = "left_leg_mc"), ("right_leg" = "right_leg_mc"), ("torso" = "torso_mc"), ("right_arm" = "right_arm_mc"), ("left_arm" = "left_arm_mc"), ("imu" = "head_inertial_hardware_device"), ("waist_imu" = "waist_inertial_hardware_device"), ("left_arm_ft_sensor" = "ergocub_left_arm_ft"), ("right_arm_ft_sensor" = "ergocub_right_arm_ft"), ("left_leg_ft_sensor" = "left_leg-FT_remapper"), ("right_leg_ft_sensor" = "right_leg-FT_remapper")]
[ERROR] Target device left_leg-FT_remapper (network = left_leg_ft_sensor ) does not exist.
[ERROR] Cannot run attach action on device wholebodydynamics
[INFO] All actions for action level 15 of startup phase started. Waiting for unfinished actions.
[INFO] All actions for action level 15 of startup phase finished.
[WARNING] There was some problem running actions for startup phase . Please check the log and your configuration
[INFO] startup phase finished.
[ERROR] Error in startup phase... see previous messages for more info
[WARNING] Interrupt # 1 # received.
[INFO] Interrupt received. Stopping all running threads.
[INFO] interrupt1 phase starting...
[INFO] interrupt1 phase finished.
[INFO] shutdown phase starting...
[INFO] Entering action level 2 of phase shutdown
[INFO] Executing detach action, level 2 on device wholebodydynamics with parameters []
[ERROR] yarprobotinterface intercepted a segmentation fault caused by a faulty plugin:
[ERROR] /home/isorrentino/dev/robotology-superbuild/build/install/lib/yarp/yarp_controlboardremapper.so(+0x3f710) [0x7f264823b710]
Trace requested at /home/isorrentino/dev/robotology-superbuild/src/YARP/src/yarprobotinterface/Module.cpp:77 by code called from:
/home/isorrentino/dev/robotology-superbuild/build/install/lib/libYARP_os.so.3(_Z16yarp_print_traceP8_IO_FILEPKcj+0x3b) [0x7f26627e7b0b]
yarprobotinterface(+0x9946) [0x561e8107f946]
/lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f2662363090]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/yarp/yarp_controlboardremapper.so(+0x3f710) [0x7f264823b710]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/yarp/wholeBodyDynamicsDevice.so(_ZN4yarp3dev23WholeBodyDynamicsDevice24resetGravityCompensationEv+0x49) [0x7f26484d5659]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/yarp/wholeBodyDynamicsDevice.so(_ZN4yarp3dev23WholeBodyDynamicsDevice9detachAllEv+0x54) [0x7f26484d76b4]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/libYARP_robotinterface.so.3(_ZNK4yarp14robotinterface6Device6detachEv+0x231) [0x7f266294d501]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/libYARP_robotinterface.so.3(_ZN4yarp14robotinterface5Robot7Private6detachERKNS0_6DeviceERKSt6vectorINS0_5ParamESaIS7_EE+0x142) [0x7f2662953072]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/libYARP_robotinterface.so.3(_ZN4yarp14robotinterface5Robot10enterPhaseENS0_11ActionPhaseE+0x7f6) [0x7f26629556c6]
yarprobotinterface(+0xb093) [0x561e81081093]
yarprobotinterface(+0xaa46) [0x561e81080a46]
/home/isorrentino/dev/robotology-superbuild/build/install/lib/libYARP_os.so.3(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x77) [0x7f2662843207]
yarprobotinterface(+0x8b90) [0x561e8107eb90]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f2662344083]
yarprobotinterface(+0x8c5e) [0x561e8107ec5e]
Segmentation fault (core dumped)

```

I noticed that this commit https://github.com/icub-tech-iit/ergocub-software/commit/f15ddc790b3a24bf4dd8ff5ad05583296e1aa162 starts to use the remappers for the leg FTs.

GiulioRomualdi, 2023-07-19:
cc @Nicogene @traversaro

traversaro, 2023-07-19:
Which version of YARP are you using?

isorrentino, 2023-07-19:
> Which version of YARP are you using?

3.8, update to May 16

> Which configuration file are you using for end? It seems different from the one that @Nicogene updated in his PR.

Which ones are you referring to? I use the ones of the latest ergocub software release https://github.com/icub-tech-iit/ergocub-software/releases/tag/v0.3.3

closed 2023-07-24 by Nicogene