mesh-iit/ergocub-software #156: Change/Uniform names of YARP devices wrt WBD

pattacini, 2023-07-31:
Just as follow up comment,  to avoid confusion related to we may consider change/uniform the names of YARP devices between:
* yarprobotinterface of the simulated robot
* yarprobotinterface of the wbd running outside of gazebo 

At the moment `ergocub_left_leg_ft` is:
* on yarprobotinterface of the simulated robot is just the name of the device that exposes the `l_leg_ft_sensor` sensor (see https://github.com/icub-tech-iit/ergocub-software/blob/facf5a613d602cb5ccb42270f248a998a0001ee8/urdf/ergoCub/conf/FT/gazebo_ergocub_left_leg_ft.ini#L2C16-L2C35)
* in yarprobotinterface of the wbd running outside of gazebo, it is the name of the `multipleanalogsensorsclient` that exposes the `l_leg_ft_sensor`, and `l_foot_rear_ft_sensor`, `l_foot_front_ft_sensor` sensors  https://github.com/icub-tech-iit/ergocub-software/blob/facf5a613d602cb5ccb42270f248a998a0001ee8/urdf/ergoCub/conf/launch_wholebodydynamics_ecub.xml#L59

Having the same name for such different devices may be source of confusion also in the future.

_Originally posted by @traversaro in https://github.com/icub-tech-iit/ergocub-software/issues/154#issuecomment-1658228905_

pattacini, 2023-07-31:
cc @Nicogene

Nicogene, 2023-09-05:
Alongside @traversaro we decided to not do this change:
- https://github.com/icub-tech-iit/ergocub-software/pull/167#issuecomment-1706192395

closed 2023-09-05 by Nicogene