mesh-iit/ergocub-software #157: Add ergoCubGazeboV1_1 ergoCubGazeboV1_1_minContacts ergoCubSN001

Nicogene, 2023-08-07:
## Task description

We would need to add files for generating the urdf of:
- `ergoCubGazeboV1_1`
- `ergoCubGazeboV1_1_minContacts`
- `ergoCubSN001`

We would need to:
- [ ] Generate meshes and simmechanics xml of ergoCub 1.1
- [ ] Create the yaml files for handling the new robot.

⚠️ Note that ergoCub 1.0 will never update regenerating from cad using the simmechanics_to_urdf procedure, so no changes are needed for it.

## Definition of Done

The above list is completed

Nicogene, 2023-08-09:
Alongside @mfussi66 today we made a plan for adding the configuration files of ergoCub 1.1.
The idea is to have under `urdf/simmechanics` three folders, `ergocub` and `ergocub1_1` and `conf_common`.
Inside `conf_common` we will put all the configuration files that are common between different versions of robots such as the ini/xml files of YARP plugins.
At configuration phase the configuration files contained in `conf_common` and `ergocubx_y/conf` will be merged in `conf_x_y`.

lrapetti, 2023-08-24:
> * ergoCubGazeboV1_1_minContacts

Hi @Nicogene, I might have lost conversations going on in the last period. I was curious which contacts will be included in this model?

Nicogene, 2023-08-24:
> > * ergoCubGazeboV1_1_minContacts
> 
> Hi @Nicogene, I might have lost conversations going on in the last period. I was curious which contacts will be included in this model?

This model have been added by @SimoneMic, you can see the changes in these PRs:
- https://github.com/icub-tech-iit/ergocub-software/pull/83
- https://github.com/icub-tech-iit/ergocub-software/pull/92

If I recall correctly, that model has contacts only in the soles, it used for navigation purposes, for increasing the performance in simulation

SimoneMic, 2023-08-24:
> If I recall correctly, that model has contacts only in the soles, it used for navigation purposes, for increasing the performance in simulation

Yes, it's done to speed up the simulation in Gazebo to avoid the computation of the collisions between the robot components with the environment. Naturally it's useful mainly in navigation, when the robot doesn't need to interact with the environment.

Nicogene, 2023-08-24:
Since thanks to https://github.com/icub-tech-iit/creo2urdf/pull/44 `creo2urdf` is capable of producing a complete ergoCub urdf, the ergocub 1.1 urdf will be added without passing through simmechanics_to_urdf.

This will require a restructure of this repository

cc @pattacini @mfussi66 @traversaro

Nicogene, 2023-08-24:
I started working on this PR:

- https://github.com/icub-tech-iit/ergocub-software/pull/161

Nicogene, 2023-08-25:
In https://github.com/icub-tech-iit/ergocub-software/commit/df1987be358f8c8525f92c9684a82bbb75319ece we have added the yaml and urdf that are passing all the unit tests

```
    Start 1: ergoCubGazeboV1ConsistencyCheck
1/4 Test #1: ergoCubGazeboV1ConsistencyCheck .....   Passed    0.03 sec
    Start 2: ergoCubGazeboV1_1ConsistencyCheck
2/4 Test #2: ergoCubGazeboV1_1ConsistencyCheck ...   Passed    0.03 sec
    Start 3: ergoCubSN000ConsistencyCheck
3/4 Test #3: ergoCubSN000ConsistencyCheck ........   Passed    0.03 sec
    Start 4: ergoCubSN001ConsistencyCheck
4/4 Test #4: ergoCubSN001ConsistencyCheck ........   Passed    0.03 sec
```

Once we finish implementing the collisionGeometry part we can define this activity done.

closed 2023-08-29 by Nicogene

Nicogene, 2023-08-29:
Since the coupling handler for hand mk5.1 is not yet available in gazebo-yarp-plugins, ergoCub 1.1 urdf is using the "old" coupling mk5, even if it has hands mk5.1.

This will be fixed as soon as the new coupling handler is added. Note that until that time the performance in simulation of the hands **will not** reflect the performances of the real ones.

cc @xEnVrE @traversaro @SimoneMic @GiulioRomualdi @lrapetti @pattacini @fiorisi @Lawproto @maggia80 @fedeceola @Woolfrey @vigisushrutha23 @gabrielecaddeo

Nicogene, 2023-09-13:
An update on this, the work on coupling laws of mk5.1 has been completed, see

- https://github.com/robotology/gazebo-yarp-plugins/issues/661

PR #161 (2023-08-29): Refactor the repo for using creo2urdf instead of simmechanics
This PR refactors the structure of the repository for not using anymore `simmechanics_to_urdf` for generating the urdf but [`creo2urdf`](https://github.com/icub-tech-iit/creo2urdf).
For the urdf part, it hosts only the yaml used by the plugin and the urdf itself.

Moreover, it adds:
- ergoCubGazeboV1_1
- ergoCubGazeboV1_1_minContacts
- ergoCubSN001

It will be probably a first iteration, right now the configuration files (xml) are shared between robots, and there are duplications in yamls

It fixes #157