mesh-iit/ergocub-software #168: Discrepancy between documented and implemented limits for ergoCub hand MK5

xEnVrE, 2023-09-07:
After the recent update of the documentation in https://icub-tech-iit.github.io/documentation/hands/hands_mk5/#mk50, I noticed that there might be problems with some of the limits.

For example, the thumb proximal has 90 degrees as limit:

[image]

The current configuration file, instead (check the second column)

https://github.com/icub-tech-iit/ergocub-software/blob/5ca85d3b065f47ea652bb413d9b0536f5dfcc72c/urdf/ergoCub/conf/gazebo_ergocub_left_hand.ini#L13

has 82.1 degrees.

Although we could just update the configuration file with the documentation, I am not really sure whether they are correct. Indeed, I remember, while looking at it in the CAD with @Lawproto, that the thumb proximal could not really reach 90 degrees but a little less.

cc @Lawproto @traversaro

Lawproto, 2023-09-07:
@xEnVrE you are right, I confirm.

[image]

Lawproto, 2023-09-07:
@xEnVrE pull request opened. 
https://github.com/icub-tech-iit/documentation/pull/304

xEnVrE, 2023-09-07:
Thanks @Lawproto

Lawproto, 2023-09-07:
@xEnVrE you are welcome. The pull request was merged, now you can see the correct value in the documentation.

[image]

xEnVrE, 2023-09-07:
Thanks @Lawproto.

As discussed f2f with @Lawproto, when the PR was opened I did not already notice that also the `l_thumb_dist` is different between the two sources:
- `53.6` within the config file
- `59.2` within the documentation

Would it be possible to also verify this?

**Note: all the other numbers are correct!**

Lawproto, 2023-09-08:
@xEnVrE from the CAD model, I say the config file is right.

[image]

I opened ad PR: https://github.com/icub-tech-iit/documentation/pull/305.

xEnVrE, 2023-09-08:
Thanks heaps @Lawproto

xEnVrE, 2023-09-08:
Closing as the issue has been solved.

closed 2023-09-08 by xEnVrE