mesh-iit/ergocub-software #170: Automatic indent urdf file after generation

GiulioRomualdi, 2023-09-11:
## Task description
It would be nice to automatically indent the urdf file at the end of the generation. This will simplify the manual inspection of the model of the robot from file.

## Definition of Done
A procedure to automatically indent the file once generated (or at the end of the generation)

cc @traversaro

traversaro, 2023-09-11:
Probably we can just run one of the following command as part of the generation process: https://stackoverflow.com/questions/16090869/how-to-pretty-print-xml-from-the-command-line .

traversaro, 2023-09-11:
If we want to do that as part of creo2urdf/idyntree, the function used by `xmllint --format` is `xmlSaveFormatFileEnc`, see https://github.com/GNOME/libxml2/blob/99cba4b37b1a39d477f1b5acf782c6bd2d16d0ee/xmllint.c#L2619C10-L2619C30 .

Nicogene, 2023-09-12:
Looking at the [`libxml2` doc](https://gnome.pages.gitlab.gnome.org/libxml2/devhelp/libxml2-tree.html) I found this solution.

- https://github.com/icub-tech-iit/creo2urdf/pull/54

See the resulting urdfs:

- https://github.com/icub-tech-iit/ergocub-software/pull/171

PR #171 (2023-09-12): ergoCubV1_1: fix identation
As per title

It fixes #170 

cc @GiulioRomualdi @traversaro
closed 2023-09-12 by Nicogene