mesh-iit/ergocub-software #172: Clarification on the zero configuration of *_index_add of the simulated Mk5.1 hand

xEnVrE, 2023-09-13:
I am opening this to ask a clarification about the behavior of the `*_index_add` joint of the Mk5 **.1** hand.

According to [this](https://github.com/icub-tech-iit/ergocub-design-lowerarm/issues/157#issuecomment-1660566060) comment, it was decided that the zero position of the index adduction is the one where the index is more distant from the other fingers, i.e.:

[image]

It seems that such new behavior has not been transferred correctly in the simulation:

### index_add = 0

[image]

### index_add = 15

[image]

I guess we should:
- invert the direction of the axis of rotation
- change the default configuration at startup in Gazebo

cc @Nicogene @pattacini @fiorisi

pattacini, 2023-09-14:
Good catch @xEnVrE 👍🏻 

@Nicogene @mfussi66 please read up on the issue from https://github.com/icub-tech-iit/ergocub-design-lowerarm/issues/157#issuecomment-1660679203 onward for more info on that.

You were off, I tried to find out why we ended up with a different convention wrt the iCub hand but unsuccessfully. At this point, we ought to conform to the latest convention.

Nicogene, 2023-09-14:
Since creo2urdf assumes that in the simulation model, all the joints are in their zeros, I need that @Mick3Lozzo or @Lawproto to change the rest position of the `index_add` joints in the simulation model.

What about the `thumb_add` ? Also that has to be changed?

xEnVrE, 2023-09-14:
> Or actually, maybe it is the chance for changing the range to -3 to 12 degrees

I am a bit confused, isn't it an issue related to Mk5.0?

> What about the thumb_add ? Also that has to be changed?

I would say no, but others might comment on this. The behavior of the thumb adduction is at the moment similar to that on the old iCub hands, i.e. 0.0 -> thumb within the plane of the palm, max -> thumb ~orthogonal to the palm or the like

pattacini, 2023-09-14:
>Since creo2urdf assumes that in the simulation model, all the joints are in their zeros, I need that @Mick3Lozzo or @Lawproto to change the rest position of the index_add joints in the simulation model.

Let's do it.

Lawproto, 2023-09-14:
@Nicogene @pattacini to sum it up, I inderstood I have to update the ergoCub 1.1 sim model with the new index position.

[image]

It this is the case, the `index_add` joint rom spans from 0 (figure) to 15°.

pattacini, 2023-09-14:
Yep 👍🏻

Nicogene, 2023-09-15:
After @Lawproto fix, here is the PR that port those changes in the urdf:

- https://github.com/icub-tech-iit/ergocub-software/pull/173

### 0 deg
[image]
### 15 deg
[image]

cc @fiorisi @pattacini @xEnVrE

PR #173 (2023-09-15): ergoCubV1_1: correct index_add joint
It fixes #172
closed 2023-09-15 by Nicogene