mesh-iit/ergocub-software #174: [ergoCubGazeboV1*]: wrong inertial values on the feet

martinaxgloria, 2023-09-19:
Trying to import ergoCubGazeboV1* models on gazebo, I spotted that the model was unstable, it was like wobbling on the feet. Together with @Nicogene, we used git bisect to understand which commit introduced the bug and it was the https://github.com/icub-tech-iit/ergocub-software/commit/aae9cd689de40f900512fc1ad6a9c0c9b110d02b. We verified that the bug was not due to the introduction of the `ft frames` but the changes in the inertial values (actually some values have a different sign). 

Any idea on how to fix this problem?

cc  @traversaro

traversaro, 2023-09-19:
Which version of Gazebo you are using?

traversaro, 2023-09-20:
> We verified that the bug was not due to the introduction of the ft frames but the changes in the inertial values, actually some values have a different sign (see for example [here](https://github.com/icub-tech-iit/ergocub-software/pull/159/files#diff-f79a1cb254f3eadc85bf53d10525546464638b7a666ab54493dbe7721b3f1a5dR255))

Related PR: https://github.com/icub-tech-iit/ergocub-software/pull/158 .

traversaro, 2023-09-20:
> > We verified that the bug was not due to the introduction of the ft frames but the changes in the inertial values, actually some values have a different sign (see for example [here](https://github.com/icub-tech-iit/ergocub-software/pull/159/files#diff-f79a1cb254f3eadc85bf53d10525546464638b7a666ab54493dbe7721b3f1a5dR255))
> 
> Good catch! Let me think a bit about it. For sure we could add a check on the physical consistency of the inertial parameters in the CI, by using the [`iDynTree::SpatialInertia::isPhysicallyConsistent()`](https://robotology.github.io/idyntree/classiDynTree_1_1SpatialInertia.html#a6b493d81ed9bcedb37f81d0125d1ab04) method

Actually that is not the problem: the inertial parameter that changed sign is the com Z, and that make sense as it is expressed in the link frame, and we moved the link frame up, so the COM w.r.t. the link frame is now negative. However, it is not clear how this influences gazebo.

traversaro, 2023-09-20:
I do not think the problem is in the COM location, by looking into the model, it seems that the collision of the links `r_foot_rear`, `r_foot_front`,  `l_foot_rear`, `l_foot_front` are wrongly placed:

[image]

traversaro, 2023-09-20:
And for completeness: the problem is that `r_ankle_2` and `l_ankle_2` bodies are not simple geometries, and we do not set their contact parameters in the .yaml file.

traversaro, 2023-09-20:
I proposed a solution in https://github.com/icub-tech-iit/ergocub-software/pull/175, the PR also contains a (hopefully) detailed description of what happened, please comment if it is not clear. 

Unfortunatly https://github.com/icub-tech-iit/ergocub-software/pull/161 removed the simmechanics generation machinery, so to propose the change I created a `ergocub-0.3` branch out of `v0.3.4` and I opened the PR there. @Nicogene feel free to decide what is the best option to propagate this modifications in the master branch.

martinaxgloria, 2023-09-20:
Thank you @traversaro!