mesh-iit/ergocub-software #179: Use simple collision for foot also for non-Gazebo models

traversaro, 2023-09-28:
With @akhilsathuluri we are doing physically realistic simulations of ergoCubSN000 and ergoSN001 (we want to use the real inertia of the robot, and not the irrealistic inertia in ergoCubGazebo models) for doing so, it would be convenient to have the collision shape of the foot as a simple shape also there. Do you think we can have the simple shape of the foot also there for ergoCubSN000 and ergoSN001? 

fyi @Nicogene @S-Dafarra

pattacini, 2023-09-29:
/remind next Tuesday cc @Nicogene

Nicogene, 2023-10-03:
> you think we can have the simple shape of the foot also there for ergoCubSN000 and ergoSN001?

It is ok for me 👍🏻

Nicogene, 2023-10-03:
Updating ergoCubSN000 is not possible using `creo2urdf` due to [this issue](https://github.com/icub-tech-iit/creo2urdf/issues/40#issuecomment-1745141839) we decided to fix it only in ergoCubSN001.

If ergoCubSN001 is not sufficient I will try to generate the ergoCubSN000 w/ `simmechanics_to_urdf`

PR #180 (2023-10-04): ergoCubSN001: simplify collision shape of feet
It fixes #179 

Moreover, it removes all the `default` sensors from the ergoCubSN000 yaml, since that clause is not supported by `creo2urdf`
closed 2023-10-04 by Nicogene