mesh-iit/ergocub-software #184: `ErgoCubGazeboV1_minContacts`: Collisions positions

flferretti, 2023-10-26:
## Task description

When working with `ErgoCubGazeboV1_minContacts` and `ErgoCubGazeboV1_1_minContacts` downloaded from the robotology channel with `conda`, I noticed that the box collisions on the feet were not below or matching the soles as expected. (See image below)

[image]

I was then wondering if this were the expected result or if the offset needs to be changed.

C.C. @lrapetti @CarlottaSartore

traversaro, 2023-10-26:
Which version of ergocub software did you install?

traversaro, 2023-10-26:
This seems similar to https://github.com/icub-tech-iit/ergocub-software/issues/174 , that was fixed in master, perhaps we could do a release of ergocub-software to release the fix? @Nicogene

flferretti, 2023-10-26:
> This seems similar to #174 , that was fixed in master, perhaps we could do a release of ergocub-software to release the fix? @Nicogene

Yes, in the latest URDF version the problem has been solved. Thanks

Nicogene, 2023-10-26:
Release done:
- https://github.com/icub-tech-iit/ergocub-software/releases/tag/v0.5.0

flferretti, 2023-10-26:
Thank you all, closing

closed 2023-10-26 by flferretti

traversaro, 2023-10-26:
Thanks! Unfortunately at the moment ergocub-software is not on conda-forge so it will take a few days to get the new package in the robotology channel.

traversaro, 2023-10-31:
Update packages are available in robotology channel: https://anaconda.org/robotology/ergocub-software  (except for Windows that have some problems).