mesh-iit/ergocub-software #186: Misalignment of the l_sole and r_sole frames in ergoCubSN001 and ergoCubGazeboV1_1 models

GiulioRomualdi, 2023-11-03:
## Task description
Upon examining the `l_sole` and `r_sole` frames in both `ergoCubSN001` and `ergoCubGazeboV1_1`, we have observed that they are not properly located on the robot's sole.

In this visualization, I am plotting the `r_sole` and `l_sole` frames. On the left, we have `ergoCubSN000`, and on the right, we have `ergoCubSN001`. The frames appear to be centered in different positions, with the frames for `ergoCubSN001` appearing higher.

[image]

## Definition of Done
Our expectation was that l_sole and r_sole would be positioned in the same manner as in ergoCubSN000.

GiulioRomualdi, 2023-11-03:
Applying this patch to the robot model the frames are placed as expected

```patch
diff --git a/urdf/ergoCub/robots/ergoCubGazeboV1_1/model.urdf b/urdf/ergoCub/robots/ergoCubGazeboV1_1/model.urdf
index 3430d58..b7826c9 100644
--- a/urdf/ergoCub/robots/ergoCubGazeboV1_1/model.urdf
+++ b/urdf/ergoCub/robots/ergoCubGazeboV1_1/model.urdf
@@ -2508,7 +2508,7 @@
   </joint>
   <link name="l_sole"/>
   <joint name="l_sole_fixed_joint" type="fixed">
-    <origin xyz="0.059629999999999996 0 1.1102230246251565e-16" rpy="0 1.5094274020956558e-31 0"/>
+    <origin xyz="0.059629999999999996 0 -0.022299999999999986" rpy="0 1.5094274020956558e-31 0"/>^M
     <parent link="l_foot_rear"/>
     <child link="l_sole"/>
   </joint>
@@ -2874,7 +2874,7 @@
   </joint>
   <link name="r_sole"/>
   <joint name="r_sole_fixed_joint" type="fixed">
-    <origin xyz="0.05963 -1.3877787807814457e-17 7.771561172376096e-16" rpy="0 3.1371750050201436e-30 0"/>
+    <origin xyz="0.05963 -1.3877787807814457e-17 -0.022299999999999986" rpy="0 3.1371750050201436e-30 0"/>^M
     <parent link="r_foot_rear"/>
     <child link="r_sole"/>
   </joint>
```

GiulioRomualdi, 2023-11-03:
To be more specific the definition of done is to be compliant with https://www.ros.org/reps/rep-0120.html#l-sole-and-r-sole
> [l_sole and r_sole](https://www.ros.org/reps/rep-0120.html#id12)
> The coordinate frame called l_sole defines the position and orientation of the left leg support body. The (X,Y) plane of this frame must be coplanar with the support body contact surface. It is recommended to choose the origin as the projection of the support body origin on the contact surface. The Z coordinate of this frame should be on the ground level when standing.
> 
> The coordinate frame called r_sole defines the equivalent frame for the right leg.

Nicogene, 2023-11-03:
To give a context here, for this issue:
- https://github.com/icub-tech-iit/ergocub-software/issues/146

We moved up the frames of the parts in order to be coincident w/ the ones of the ft sensor. Probably `l/r_sole` has been moved up by mistake

Mick3Lozzo, 2023-11-06:
[image]

Yes, `L_SOLE` and `R_SOLE` were defined following L_FOOT frames. I am going to correct them.

@Nicogene @GiulioRomualdi

Mick3Lozzo, 2023-11-06:
@GiulioRomualdi as per [this comment](https://github.com/ami-iit/component_bipedal-locomotion/issues/100#issuecomment-1792719024) I verified that `L_SOLE` and `R_SOLE` are 2 mm higher than the contact surface, that happens on the rubber soles.

[image]

[image]

Do you mean that they should be located on the lower part of the rubber soles, that is the effective contact surface?

@Nicogene

Mick3Lozzo, 2023-11-06:
Updated ergoCub 1.1  CAD and SIM shrinkwrapped model.

@Nicogene @GiulioRomualdi

commit ad435d59 (2023-12-13): ergoCubV1_1: move down the l/r_sole frames

It fixes #186