mesh-iit/ergocub-software #190: Cannot reset the robot in simulation when using DART

PasMarra, 2023-11-13:
As per title, I cannot reset ergoCub in simulation when using DART. Some joints go in hardware fault when I try.

Nicogene, 2023-11-15:
Hi @PasMarra, could you please post which error/misbehavior you are experiencing?
What do you mean for "reset"?
"Using DART" it means using DART physical engine in gazebo? Which version of gazebo you are using? Gazebo classic or the new one?

PasMarra, 2023-11-15:
Hi @Nicogene, here you can find the required info:

> could you please post which error you are experiencing? (the cause of hardware fault)
> What do you mean for "reset"?

This is what happen when I reset the world (ctrl+R):

https://github.com/icub-tech-iit/ergocub-software/assets/143645988/ebddf160-e192-465b-85f5-9435cee36255

```sh
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| INITIAL CONFIGURATION IS:  -0.520000	 0.520000	 0.000000	 0.785000	 0.000000	 0.000000	 0.000000
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with default values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| INITIAL CONFIGURATION IS:  -0.520000	 0.520000	 0.000000	 0.785000	 0.000000	 0.000000	 0.000000
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with default values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  1  torque too big! (  -2030.98  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  4 torque too big! (  -2012.41  )

```

> "Using DART" it means using DART physical engine in gazebo?

Yes, indeed.
```
gazebo -e dart
```
> Using other engines fixes this problem?

Yes, it does. For example, with ODE there isn't any problem.

> Which version of gazebo you are using?
> Gazebo classic or the new one?

Gazebo classic, version 11.10.2

> Which ergoCub model are you using?

ergocubGazeboV1_1

> Which version of ergocub-software?

`dea966a64071bacff91fdac1919003877cf7d9f5`

xEnVrE, 2023-11-15:
> Using other engines fixes this problem?

Let me add that although using `ode` the problem is not there, we would like to use `DART` as it works best for our scenario (e.g. objects not drifting on the table, better contacts). Thanks

Nicogene, 2023-11-16:
To me is not clear what happens on the gyp plugins when you reset the world, it seems they are not reloaded, but they do something since I am seeing this log

```
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| INITIAL CONFIGURATION IS:  -0.520000        0.520000        0.000000        0.785000        0.000000        0.000000        0.000000
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with default values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| INITIAL CONFIGURATION IS:  -0.520000        0.520000        0.000000        0.785000        0.000000        0.000000        0.000000
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with default values
[DEBUG] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| Initializing Trajectory Generator with current values
```

It seems it is commanding to the controlboards the initial configuration of the robot

Maybe @traversaro or @randaz81 knows something about it?

I am not understanding why this HF does not happen when you insert the model, since the execution seems the same

Nicogene, 2023-11-16:
I tried to set `max_torques` to 9999.9 and still we are getting the hw fault

```
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  1  torque too big! (  -10005.7  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  0  torque too big! (  -10015.4  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  0  torque too big! (  -10017.7  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  4  torque too big! (  10009.9  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  4  torque too big! (  10009.8  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  3  torque too big! (  -10002  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  3  torque too big! (  -10001.2  )
[ERROR] |gazebo-yarp-plugins.plugins.GazeboYarpControlBoard| An hardware fault occurred on joint  0  torque too big! (  10008.2  )
```
So I think increasing the `max_torques` is not the right way to fix it

Nicogene, 2023-11-16:
I found this that is interesting:
- https://github.com/robotology/gazebo-yarp-plugins/issues/92
- https://github.com/robotology/gazebo-yarp-plugins/pull/247
- https://github.com/robotology/gazebo-yarp-plugins/pull/345

cc @traversaro

traversaro, 2023-11-16:
> Maybe @traversaro or @randaz81 knows something about it?

As you found yourself in https://github.com/robotology/gazebo-yarp-plugins/pull/247/files, on reset the `onReset` callback of the plugin is called. It is not clear to me anyhow what is happening. When you launch on a clean world the robot, do you see anything like that?

traversaro, 2023-11-16:
> I am not understanding why this HF does not happen when you insert the model, since the execution seems the same

Sorry, I missed this, this answer one of my previous questions.

traversaro, 2023-11-16:
> So I think increasing the `max_torques` is not the right way to fix it

That's true, but just as a workaround I would try to set max_torques to a really high value (like 1000000) to see if the simulation works fine.

Nicogene, 2023-11-16:
> > So I think increasing the `max_torques` is not the right way to fix it
> 
> That's true, but just as a workaround I would try to set max_torques to a really high value (like 1000000) to see if the simulation works fine.

With `100000.0` the reset works fine, @traversaro @xEnVrE @PasMarra I can open a PR w/ this very high value as `max_torques` but I think that this issue should be tracked in GYP

Nicogene, 2023-11-16:
I paused the simulation in order to check what happens when you insert the model and when you reset the world and seems that it is falling from the very same height:

|  Put in the model |  Reset world  | 
|---|---|
|  [image] | [image]  |  

Unfortunately this https://github.com/icub-tech-iit/ergocub-software/issues/190#issuecomment-1814406543 is not enough, some times goes in HF also with these high threshold, I can increase it more in case

traversaro, 2023-11-16:
This is kind of ortogonal, but why don't we set the robot height such that it does not jump at the beginning of the simulation?

Nicogene, 2023-11-16:
> This is kind of ortogonal, but why don't we set the robot height such that it does not jump at the beginning of the simulation?

If you mean to lower the spawn point I tried to set it at 0.78 m height that is the lower possible since the root is 0.779 meters from the sole plane, but this problem persist

traversaro, 2023-11-16:
> If you mean to lower the spawn point I tried to set it at 0.78 m height that is the lower possible since the root is 0.779 meters from the sole plane, but this problem persist

Ok, thanks! Anyhow it probably make sense to implement that change anyhow, independently by this problem?

Nicogene, 2023-11-22:
@martinaxgloria noticed that this problems happens also w/ `iCubGazeboV2_5`

Nicogene, 2023-12-05:
> > @martinaxgloria noticed that this problems happens also w/ `iCubGazeboV2_5`
> 
> Is this tracked or solved somewhere?

Opened now:

- https://github.com/robotology/icub-models/issues/224

traversaro, 2023-12-05:
> > > @martinaxgloria noticed that this problems happens also w/ `iCubGazeboV2_5`
> > 
> > Is this tracked or solved somewhere?
> 
> Opened now:
> 
> - https://github.com/robotology/icub-models/issues/224

Thanks!

PasMarra, 2023-12-06:
https://github.com/icub-tech-iit/ergocub-software/assets/143645988/ac16889e-c35e-409f-a450-01e81b625566

Hi @Nicogene, as you can see from the video, incrementing `max_torques` to `1e6` is not sufficient for a successful reset.

ergocub-software version `0945da4e405e72b6fa45664a027d0d6fe0fcb008`

cc: @traversaro @xEnVrE

traversaro, 2023-12-06:
While the problem is indeed there, @PasMarra for your experiments have you considered either to fix the the robot base to the world, or at least to create a world in which the model is already placed of the correct location, instead of spawning it far from the ground?

commit 6d3ff143 (2023-12-04): ergoCubV1*: increase the max_torques to 1000000

It fixes #190